SpikeInterface/spikeextractors #320: error converting from OpenEphys to NWB

bendichter, 2020-02-18:
It looks like there is an issue converting from OpenEphys to NWB when location information is missing:

```python
from spikeextractors import OpenEphysRecordingExtractor, NwbRecordingExtractor

fpath = '/Volumes/GoogleDrive/Shared drives/ePhys_Processing_Soltesz/JF_298_003_2020-02-04_15-15-40'
re = OpenEphysRecordingExtractor(fpath)
out_path = '/Volumes/GoogleDrive/My Drive/data/Soltesz/test.nwb'
NwbRecordingExtractor.write_recording(re, out_path)
```

```
Reading all channels
100_CH1.continuous
100_CH2.continuous
100_CH3.continuous
100_CH4.continuous
100_CH5.continuous
100_CH6.continuous
100_CH7.continuous
100_CH8.continuous
100_CH9.continuous
100_CH10.continuous
100_CH11.continuous
100_CH12.continuous
100_CH13.continuous
100_CH14.continuous
100_CH15.continuous
100_CH16.continuous
100_CH17.continuous
100_CH18.continuous
100_CH19.continuous
100_CH20.continuous
100_CH21.continuous
100_CH22.continuous
100_CH23.continuous
100_CH24.continuous
100_CH25.continuous
100_CH26.continuous
100_CH27.continuous
100_CH28.continuous
100_CH29.continuous
100_CH30.continuous
100_CH31.continuous
100_CH32.continuous

/Users/bendichter/opt/anaconda3/lib/python3.7/site-packages/pynwb/file.py:694: UserWarning: Date is missing timezone information. Updating to local timezone.
  warn("Date is missing timezone information. Updating to local timezone.")
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-8-5bba4172f62d> in <module>
      1 out_path = '/Volumes/GoogleDrive/My Drive/data/Soltesz/test.nwb'
----> 2 NwbRecordingExtractor.write_recording(re, out_path)

~/opt/anaconda3/lib/python3.7/site-packages/spikeextractors/extractors/nwbextractors/nwbextractors.py in write_recording(recording, save_path, acquisition_name, **nwbfile_kwargs)
    320             for m in channel_ids:
    321                 if m not in nwb_elec_ids:
--> 322                     location = recording.get_channel_property(m, 'location')
    323                     while len(location) < 2:
    324                         location = np.append(location, [0])

~/opt/anaconda3/lib/python3.7/site-packages/spikeextractors/recordingextractor.py in get_channel_property(self, channel_id, property_name)
    393             raise ValueError('no properties found for channel' + str(channel_id))
    394         if property_name not in self._channel_properties[channel_id]:
--> 395             raise RuntimeError(str(property_name) + " has not been added to channel " + str(channel_id))
    396         if not isinstance(property_name, str):
    397             raise TypeError(str(property_name) + " must be a string")

RuntimeError: location has not been added to channel 0
```

Similar error when requesting location from the `OpenEphysRecordingExtractor` object:

```python
re.get_channel_locations()
```

```python
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-11-455d3dd614d7> in <module>
----> 1 re.get_channel_locations()

~/opt/anaconda3/lib/python3.7/site-packages/spikeextractors/recordingextractor.py in get_channel_locations(self, channel_ids)
    244         locations = []
    245         for channel_id in channel_ids:
--> 246             location = self.get_channel_property(channel_id, 'location')
    247             locations.append(location)
    248         return locations

~/opt/anaconda3/lib/python3.7/site-packages/spikeextractors/recordingextractor.py in get_channel_property(self, channel_id, property_name)
    393             raise ValueError('no properties found for channel' + str(channel_id))
    394         if property_name not in self._channel_properties[channel_id]:
--> 395             raise RuntimeError(str(property_name) + " has not been added to channel " + str(channel_id))
    396         if not isinstance(property_name, str):
    397             raise TypeError(str(property_name) + " must be a string")

RuntimeError: location has not been added to channel 0
```

bendichter, 2020-02-18:
@luiztauffer

luiztauffer, 2020-03-26:
I think this should be fixed with the changes in #309 
can you share a file so I can test it?

PR #309 (2020-03-27): Nwb refactor metadata
This refactor contains some changes that help us integrate NwbRecordingExtractor better with many of our current projects, plus some fixes:

- add metadata control to NwbExtractor. This allows the use of metadata file instructions for NWB file contruction, a practice we’ve been trying to promote. It is optional and the standard conversions recording <--> nwb should work just the same without it.

- breaks down the nwb file formatting code into smaller, nwb-class-specific static methods, which allows us to interact better with each step of the way if needed for more advanc [... cut]
closed 2020-03-27 by alejoe91