SpikeInterface/spikeextractors #323: Master branch failing NWB test despite the merged PR passing

colehurwitz, 2020-02-22:
It seems that the NWB test is failing for some reason on master branch (despite the PR I merged passing the same test). Could you take a look? @bendichter @luiztauffer 

```
self = <test_extractors.TestExtractors testMethod=test_nwb_extractor>
    def test_nwb_extractor(self):
        path1 = self.test_dir + '/test.nwb'
        se.NwbRecordingExtractor.write_recording(self.RX, path1)
        RX_nwb = se.NwbRecordingExtractor(path1)
        self._check_recording_return_types(RX_nwb)
        self._check_recordings_equal(self.RX, RX_nwb)
        del RX_nwb
        # overwrite
        se.NwbRecordingExtractor.write_recording(self.RX, path1, session_description='second',
                                                 identifier='19475')
        RX_nwb = se.NwbRecordingExtractor(path1)
        self._check_recording_return_types(RX_nwb)
        self._check_recordings_equal(self.RX, RX_nwb)
        # add sorting to existing
        se.NwbSortingExtractor.write_sorting(self.SX, path1)
        # create new
        path2 = self.test_dir + '/firings_true.nwb'
        se.NwbSortingExtractor.write_sorting(self.SX, path2, session_description='second',
                                             identifier='19475')
        SX_nwb = se.NwbSortingExtractor(path1)
>       self._check_sortings_equal(self.SX, SX_nwb)
tests/test_extractors.py:264: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
tests/test_extractors.py:353: in _check_sortings_equal
    self.assertTrue(np.array_equal(train1, train2))
E   AssertionError: False is not true
```

alejoe91, 2020-03-06:
There is also another error now:
https://travis-ci.org/SpikeInterface/spikeextractors/builds/659091279#L467-L544

probably due to a new pynweb version? we can disable it for now @colehurwitz

colehurwitz, 2020-03-06:
Ok, that is a good idea

bendichter, 2020-03-15:
@colehurwitz Thanks for the report. We added columns rel_x and rel_y to the electrodes table. In a recent release, pynwb added these cols, so now we are adding columns that already exist. We are working on pynwb to fix this so it doesn't throw an error and we will work on this repo so that it works across versions of pynwb, including the current one.

luiztauffer, 2020-03-26:
This is the current state on the issue:

- `rel_x`, `rel_y` and `rel_z` were included in the latest pynwb release (1.3.0) as pre-existing optional arguments for the electrodes table. Therefore, for `pynwb==1.3.0` we shouldn’t create these columns, but instead simply pass them as arguments when adding electrodes to the table. The current version is problematic though (see https://github.com/NeurodataWithoutBorders/pynwb/issues/1204) and we need to wait for a fix in a next release.

- For now we can fix `pynwb<1.3.0` and, in the code, test the version before creating  `rel_x`, `rel_y` and `rel_z` columns. This should work and it’s in the current draft PR #309

bendichter, 2020-03-27:
this should be fixed now

closed 2020-03-30 by alejoe91