SpikeInterface/spikeextractors #336: Some precision in the doc for MountainSort/SpikeInterface

llobetv, 2020-03-24:
Hi first thank you for this great works, it will really make my life simpler!  

I try to use spikeinterface for mountainSort and I encounter sevral issue that could be easily be add in the doc for futur users:
1. I needed to change temp folder for mountainSort, there was no var for that, I manage to do it with thisd command : os.environ['TEMPDIR'] = "/mnt/hdd/mountainlab/mountainlab/"
2. In tutorial https://spikeinterface.readthedocs.io/en/latest/getting_started/plot_getting_started.html it take me long time to understand that recording_prb is recording + prb data and that it is this one that you have to give to spikes sorters function. It would be nice to indicate it clearly.

Best regards,
Victor Llobet

alejoe91, 2020-03-24:
Hi @llobetv 

Thanks for the interest and the support.

1. For the mountainsort4 issue, can you open it here: https://github.com/magland/ml_ms4alg? It's not related to SpikeInterface directly, but I agree that it would be nice to be able to point to a local directory.

2. Thanks for spotting it! I will modify the docs and use `recording_prb` for further steps.

Cheers,
Alessio

alejoe91, 2020-06-19:
Fixed in the docs

closed 2020-06-19 by alejoe91