SpikeInterface/spikeextractors #344: export_to_phy - BinDatRecordingExtractor - TypeError: data type "<class 'numpy.int16'>" not understood

llobetv, 2020-03-31:
Hi,
I just got this error after update spikeexctractor, it appear that dtype np.int16 doesn't work anymore for export_to_phy coupled with BinDatRecordingExtractor. I also try with "int16" and I got the same error.

Traceback (most recent call last):
  File "spikeinterface_class.py", line 132, in <module>
    si.run_ms(file_path, prb_file, ms4_params, numchan, sampling_frequency, 
  File "spikeinterface_class.py", line 94, in run_ms
    st.postprocessing.export_to_phy(recording_prb, sorting_MS4, 
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spiketoolkit-0.5.2-py3.8.egg/spiketoolkit/postprocessing/postprocessing_tools.py", line 861, in export_to_phy
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spiketoolkit-0.5.2-py3.8.egg/spiketoolkit/postprocessing/postprocessing_tools.py", line 1254, in _get_phy_data
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spiketoolkit-0.5.2-py3.8.egg/spiketoolkit/postprocessing/postprocessing_tools.py", line 1110, in _get_quality_metric_data
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spiketoolkit-0.5.2-py3.8.egg/spiketoolkit/postprocessing/postprocessing_tools.py", line 113, in get_unit_waveforms
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spiketoolkit-0.5.2-py3.8.egg/spiketoolkit/postprocessing/postprocessing_tools.py", line 1349, in _extract_waveforms_one_unit
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spikeextractors-0.7.4-py3.8.egg/spikeextractors/extraction_tools.py", line 565, in load_extractor_from_dict
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spikeextractors-0.7.4-py3.8.egg/spikeextractors/baseextractor.py", line 104, in load_extractor_from_dict
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spikeextractors-0.7.4-py3.8.egg/spikeextractors/baseextractor.py", line 121, in _load_extractor_from_dict
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spikeextractors-0.7.4-py3.8.egg/spikeextractors/baseextractor.py", line 152, in _load_extractor_from_dict
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spikeextractors-0.7.4-py3.8.egg/spikeextractors/extractors/bindatrecordingextractor/bindatrecordingextractor.py", line 32, in __init__
  File "/users/nsr/llobet/miniconda2/envs/spikeinterface/lib/python3.8/site-packages/spikeextractors-0.7.4-py3.8.egg/spikeextractors/extraction_tools.py", line 249, in read_binary
TypeError: data type "<class 'numpy.int16'>" not understood

alejoe91, 2020-03-31:
Hi @llobetv, what code are you running exactly? 
What is your extractor? I'll try to reproduce the error and fix it.
I think that the problem is that the dtype is dumped as a 'class' rather than a string.

alejoe91, 2020-03-31:
Ok I found the issue. Give me 15 min :)

alejoe91, 2020-03-31:
and thanks again for stress testing the code!!

llobetv, 2020-03-31:
Just when I ended my answer, you are too fast.

No problem, you are the one doing the job for us..., thanks :)

alejoe91, 2020-03-31:
Can you test if this fixes it? https://github.com/SpikeInterface/spikeextractors/pull/345

llobetv, 2020-03-31:
It fix the bug but now I got Bus error after several waveform extractions, I'll check that and keep you informed.

Waveform 1/238
Waveform 2/238
Waveform 3/238
Waveform 4/238
Waveform 5/238
Waveform 6/238
Bus error

llobetv, 2020-03-31:
I didn't see .raw file of waveform exctracted being created anymore

alejoe91, 2020-03-31:
HI @llobetv then I'll merge the PR. I don't think the problem is related to it.
Let's move this conversation to spiketoolkit

closed 2020-03-31 by alejoe91