SpikeInterface/spikeextractors #353: Proposal: add 'cluster_group' and 'kwik_id' properties to klustaextractors

TRuikes, 2020-04-07:
Hi!
Let me just start with that I'm not sure where to post this. If you would like me to communicate through another channel, let me know!

We are mostly working with klusta, while working with klustaextractor, I noticed the 'cluster_group' property was not available, which is the 'quality' you manually assign in the GUI (good/bad/noise). Instead the units have the 'group' property, which is the 'channel_group' property in the KWIK format. I've updated the `KlustasSortingExtractor` as shown below. With the only remark that I'm not working with `channel_groups` (only 1), so I'm not sure how this would work for multiple groups (but I think unit_ids are assigned independently `channel_groups`, which is why this should work for multiple groups as well). 

```python  
def __init__(self, file_or_folder_path):
        assert HAVE_KLSX, "To use the KlustaSortingExtractor install h5py: \n\n pip install h5py\n\n"
        SortingExtractor.__init__(self)
        kwik_file_or_folder = Path(file_or_folder_path)
        kwikfile = None
        klustafolder = None
        if kwik_file_or_folder.is_file():
            assert kwik_file_or_folder.suffix == '.kwik', "Not a '.kwik' file"
            kwikfile = Path(kwik_file_or_folder).absolute()
            klustafolder = kwikfile.parent
        elif kwik_file_or_folder.is_dir():
            klustafolder = kwik_file_or_folder
            kwikfiles = [f for f in kwik_file_or_folder.iterdir() if f.suffix == '.kwik']
            if len(kwikfiles) == 1:
                kwikfile = kwikfiles[0]
        assert kwikfile is not None, "Could not load '.kwik' file"

        kwikmodel = KwikModel(kwik_path=kwikfile.as_posix())
        self._sampling_frequency = kwikmodel.sample_rate
        self._spiketrains = []
        self._unit_ids = []
        self._kwik_ids = kwikmodel.cluster_ids
        assert len(set(self._kwik_ids)) == len(self._kwik_ids), 'duplicate kwik ids found?'

        for i, kwik_id in enumerate(self._kwik_ids):
            idx = np.nonzero(kwikmodel.spike_clusters == kwik_id)
            st = np.array(kwikmodel.spike_times)[idx]
            self._spiketrains.append(st)
            self._unit_ids.append(i)

            self.set_unit_property(i, 'cluster_group', kwikmodel.cluster_groups[kwik_id])  # noise/mua/good
            self.set_unit_property(i, 'kwik_id', kwik_id)  # usefull if in the future, i want to open gui and view unit
```

alejoe91, 2020-04-07:
Hi @TRuikes 

Thanks for reporting this. In the `PhySortingExtractor` we load the `channel_group` as `group` (which we internally use for channel groups), and the `cluster_group` as `quality`. Would that work? In that case I can implement these changes, unless you want to make a PR :)

TRuikes, 2020-04-07:
yes should work fine! thanks

TRuikes, 2020-04-08:
I've prepared a branch with an updated klustaextractor (trying to keep the original code intact). Now when I want to push my branch to origin, I'm getting an 'unable to access' error. Could you give me permission to push a branch?

alejoe91, 2020-04-08:
Hi @TRuikes 

You should make a fork and make the PR from your fork :) can you do that?

TRuikes, 2020-04-08:
right; forking. expanding my git skills daily now, thanks. I gave it a try in #354

closed 2020-04-08 by TRuikes