SpikeInterface/spikeextractors #361: PhySortingExtractor.write_sorting does not work

Dradeliomecus, 2020-04-20:
Hi,

The PhySortingExtractor does not work with my data:
```
phy_folder = "/home/aze/Documents/data"
save_folder = "/home/aze/Documents/data/test"
sorting = se.PhySortingExtractor(phy_folder)
se.PhySortingExtractor.write_sorting(sorting = sorting, save_path = save_folder)
```

At first, this doesn't work because params.py is wrong (it only contains the sampling_rate, and not the dtype or anything else. After copy pasting params.py, channels_map.npy and channels_positions.npy to the test folder, I get the following error:

```
13:54:39.831 [E] __init__:62          An error has occurred (AssertionError): 
Traceback (most recent call last):
  File "/home/aurelien/dev/miniconda3/envs/phy/bin/phy", line 8, in <module>
    sys.exit(phycli())
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/phy/apps/__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/phy/apps/template/gui.py", line 198, in template_gui
    controller = TemplateController(model=load_model(params_path), dir_path=dir_path, **kwargs)
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/phylib/io/model.py", line 1178, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/phylib/io/model.py", line 297, in __init__
    self._load_data()
  File "/home/aurelien/dev/miniconda3/envs/phy/lib/python3.7/site-packages/phylib/io/model.py", line 400, in _load_data
    self.sparse_features = self._load_features()
  File "/home/
```

alejoe91, 2020-04-20:
Hi! As shown in the docs you should use the spiketoolkit.postprocessing.export_to_phy function. I agree the write_sorting is confusing, so I'll just remove it :)

Dradeliomecus, 2020-04-20:
Oh ok, thank you!

closed 2020-04-20 by Dradeliomecus