SpikeInterface/spikeextractors #364: NeuralynxRecordingExtractor: not usable?

yarikoptic, 2020-04-20:
I am trying to try to use spikeextractors to load some data from Neuralynx . 
Sample data is available from  http://datasets.datalad.org/?dir=/labs/mvdm/MotivationalT-v2/  via DataLad or get a sample file manually by  visiting that page and getting the one you like.

With #363, if I try to provide extractor a filename or directory name - it fails with an error that too many arguments, but if I provide nothing - it still fails deeper in the neo:
```shell
$> python -c 'import spikeextractors as se; se.NeuralynxRecordingExtractor("MotivationalT-v2")'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
TypeError: __init__() takes 1 positional argument but 2 were given

$> python -c 'import spikeextractors as se; se.NeuralynxRecordingExtractor()'                  
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/yoh/proj/misc/spikeextractors/spikeextractors/extractors/neoextractors/neobaseextractor.py", line 51, in __init__
    _NeoBaseExtractor.__init__(self, **kargs)
  File "/home/yoh/proj/misc/spikeextractors/spikeextractors/extractors/neoextractors/neobaseextractor.py", line 27, in __init__
    self.neo_reader.parse_header()
  File "/home/yoh/anaconda-5.2.0-2.7/envs/mvdm-nept/lib/python3.7/site-packages/neo/rawio/baserawio.py", line 151, in parse_header
    self._parse_header()
  File "/home/yoh/anaconda-5.2.0-2.7/envs/mvdm-nept/lib/python3.7/site-packages/neo/rawio/neuralynxrawio.py", line 85, in _parse_header
    for filename in sorted(os.listdir(self.dirname)):
FileNotFoundError: [Errno 2] No such file or directory: ''
```
it is a conda environment with spikeextractors straight from git as of #363 and neo 0.8.0 installed via pip

Please advise on how to use if it is somehow a user error

alejoe91, 2020-04-20:
Hi @yarikoptic 

Probably @samuelgarcia knows what's going on!

Thanks for the other PR. 
Merging first thing tomorrow

Alessio

yarikoptic, 2020-04-20:
but in principle -- from the docstring I couldn't figure it out  from the docstring (generic abc thingie) , it seems that the issue is in the constructor way up since I should have provided some path to the extractor.

alejoe91, 2020-04-20:
It should get the same args as the neo constructor, but there night be something wrong. I agree that the docstrings should be improved! @samuelgarcia

yarikoptic, 2020-04-20:
ok, the issue is that it expects it to be a keyword argument, thus it has to be
```
se.NeuralynxRecordingExtractor(dirname=op.dirname(sample_file))
```
not just
```
se.NeuralynxRecordingExtractor(op.dirname(sample_file))
```
So it is I guess is partially a user error ;-)  ideally should be resolved by an error message which has a hint on how I screwed up and/or docstring.

Next one: 
```
se.NeuralynxRecordingExtractor(dirname=op.dirname(sample_file))
extractor = se.NeuralynxRecordingExtractor(dirname=op.dirname(sample_file))
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-6-0402e792867e> in <module>
----> 1 extractor = se.NeuralynxRecordingExtractor(dirname=op.dirname(sample_file))

~/proj/misc/spikeextractors/spikeextractors/extractors/neoextractors/neobaseextractor.py in __init__(self, **kargs)
     49     def __init__(self, **kargs):
     50         RecordingExtractor.__init__(self)
---> 51         _NeoBaseExtractor.__init__(self, **kargs)
     52 
     53         # TODO propose a meachanisim to select the appropriate channel groups

~/proj/misc/spikeextractors/spikeextractors/extractors/neoextractors/neobaseextractor.py in __init__(self, block_index, seg_index, **kargs)
     36             # auto select first segment
     37             num_seg = self.neo_reader.segment_count(block_index)
---> 38             assert num_seg == 1, 'This file is multi segment spikeextractors support only one segment'
     39             seg_index = 0
     40 

AssertionError: This file is multi segment spikeextractors support only one segment
```
which might to fall under spikeextractors umbrella -- if neo wants a directory (not just a file)... is there a way to point to a specific file? (tried  using `filename` instead of `dirname` with no success)

samuelgarcia, 2020-04-21:
Hi @yarikoptic.
Yes neo deal with folder based or file based datset. this is why you need to provide the karg (direname or filename).
We could do it automatic.

For the second error. Neo deal with multi segment multi sample rate because it render a tree of objects. In case of one AnalogSignal it is easy but when there are several objects you need to specify  ```block_index=0, seg_index=0```

I would recomand to do 
```
reader = neo.NeuralynxIO(dirname='...')
print(reader)
```
to see how much segment and channel you do have under the hood.

samuelgarcia, 2020-04-21:
Note that the wrapper neo<->spikeextractor is very simple at the moement
see https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/neoextractors/neobaseextractor.py#L29

yarikoptic, 2020-04-21:
> For the second error. Neo deal with multi segment multi sample rate because it render a tree of objects. In case of one AnalogSignal it is easy but when there are several objects you need to specify `block_index=0, seg_index=0`

thank you!  `extractor = se.NeuralynxRecordingExtractor(dirname=op.dirname(sample_file), block_index=0, seg_index=0)` worked (in a sense that it didn't crash).

Since original questions were answered and I got extractor instantiated, I will close this issue now.  But hopefully it would serve as an inspiration (and reference) to get those neo based extractors usable more easily (e.g. via better docstrings and error messages when dummy users try to use them ;))

closed 2020-04-21 by yarikoptic

samuelgarcia, 2020-04-21:
Yes of course.
Neuralynx is indeed folder based so ```dirname``` is mandatory.
Some file format like spike2 smr are file base so in that case ```filename``` is mandatory.

This neo extractor bridge is a proof of concept waiting for users.
I think you are the very first one.
I need to celebrate that by writting some docs.

yarikoptic, 2020-04-22:
> Neuralynx is indeed folder based so `dirname` is mandatory.
> Some file format like spike2 smr are file base so in that case `filename` is mandatory.

I guess underlying classes know what they need, so they could complain (well -- crash) informatively (with smth like "Neuralynx requires dirname to be provided, not filename") whenever dumb users like me use an incorrect one ;-)

samuelgarcia, 2020-04-22:
Yes but at the moment the bridge is totally generic and don't have this kind of sugar but easy add.

samuelgarcia, 2020-04-22:
BTW, you should test tridesclous

yarikoptic, 2020-04-22:
> BTW, you should test tridesclous

looks interesting indeed, you are doing great job ;) will its sorter be interfaced in spikeextractors as well?  ATM my goal is largely to establish baseline pipeline mimicking what specific lab was doing in matlab, and then formalize it into using .nwb files. I personally do not research ephys.

alejoe91, 2020-04-22:
It's already interfaced! :)

yarikoptic, 2020-04-22:
oh ;)

samuelgarcia, 2020-04-23:
We have interfaced all sorters that are benchmarked in [spikeforest](https://spikeforest.flatironinstitute.org):

kilosort (matlab + cuda)
kilosort2 (matlab  + cuda)
ironclust (matlab  + cuda)
herdingsspikes (python + c++)
spykingcircus (python + mpi)
klusta (python)
moutainsort4 (python)
tridesclous (python + opencl opentional)

You need to change only one line to change to try each then.

Have you see the [preprint](https://www.biorxiv.org/content/10.1101/796599v1) ?

[Tutorial](https://spikeinterface.readthedocs.io/en/latest/modules/index.html)  give also the big picture

[This blob](https://spikeinterface.github.io/) show several notebooks to demonstarte what can be done with the spike interface.

yarikoptic, 2020-04-23:
I think I have spotted preprint appearing but haven't read it carefully yet, thanks for all the help and references!  hopefully next time I will come up with my issues being more "informed" ;-) Cheers!