SpikeInterface/spikeextractors #367: NeuralynxRecordingExtractor().dump_to_json() leads to TypeError

yarikoptic, 2020-04-21:
Wanted to see at least some info about my new extractor (beyond default repr with an id), so tried dump_to_json:

```python
extractor = se.NeuralynxRecordingExtractor(dirname=op.dirname(sample_file), block_index=0, seg_index=0)
print(extractor.dump_to_json())
```
which lead to
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-13-8a3b1e5af5de> in <module>
      1 extractor = se.NeuralynxRecordingExtractor(dirname=op.dirname(sample_file), block_index=0, seg_index=0)
----> 2 print(extractor.dump_to_json())

~/proj/misc/spikeextractors/spikeextractors/baseextractor.py in dump_to_json(self, file_path)
     68         if self.check_if_dumpable():
     69             if file_path is None:
---> 70                 if 'Recording' in self.__class__:
     71                     file_path = 'spikeinterface_recording.json'
     72                 elif 'Sorting' in self.__class__:

TypeError: argument of type 'ABCMeta' is not iterable
```
ref: follow up to #364

alejoe91, 2020-04-21:
Can you try `extractor.dump_to_dict()`?

What's the output of `type(extractor)`?

alejoe91, 2020-04-21:
Are you able to retrieve the traces, channel_ids, sampling frequency etc.? Or does it always give this error?

alejoe91, 2020-04-21:
can you try this as well: `print(extractor.dump_to_json('test.json'))`? we might be missing a `str` there :)

yarikoptic, 2020-04-21:
thanks! providing a file name produced it and didn't crash.
```recording.dump_to_dict()``` also worked! thanks. 

I would keep this issue open, since IMHO either it should fail with descriptive exception (that filename is required) or return string json record.

alejoe91, 2020-04-21:
I think this PR https://github.com/SpikeInterface/spikeextractors/pull/370 should fix it :) Can you double check?

yarikoptic, 2020-04-21:
Checked -- both #370 and #372 fix the issue (to the degree that it doesn't crash and saves to default filename in current directory far far away from data or anything related ;))

closed 2020-04-22 by alejoe91