SpikeInterface/spikeextractors #368: 'NeuralynxRecordingExtractor' object has no attribute 'get_unit_ids'

yarikoptic, 2020-04-21:
on data from  http://datasets.datalad.org/?dir=/labs/mvdm/MotivationalT-v2/R050/R050-2014-03-31_raw

```python
sw.plot_timeseries(recording);
sw.plot_rasters(recording);
```
leads to (so plot_timeseries works, plot_rasters bugs out):
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-35-d9efd2d35931> in <module>
      1 sw.plot_timeseries(recording);
----> 2 sw.plot_rasters(recording);

~/anaconda-5.2.0-2.7/envs/mvdm-nept/lib/python3.7/site-packages/spikewidgets/widgets/rasterswidget/rasterswidget.py in plot_rasters(sorting, sampling_frequency, unit_ids, trange, color, figure, ax)
     43         color=color,
     44         figure=figure,
---> 45         ax=ax
     46     )
     47     W.plot()

~/anaconda-5.2.0-2.7/envs/mvdm-nept/lib/python3.7/site-packages/spikewidgets/widgets/rasterswidget/rasterswidget.py in __init__(self, sorting, sampling_frequency, unit_ids, trange, color, figure, ax)
     58         self._color = color
     59         self._max_frame = 0
---> 60         for unit_id in self._SX.get_unit_ids():
     61             spike_train = self._SX.get_unit_spike_train(unit_id)
     62             curr_max_frame = np.max(spike_train)

AttributeError: 'NeuralynxRecordingExtractor' object has no attribute 'get_unit_ids'
```

alejoe91, 2020-04-21:
Ok that's a recording extractor so it should not have units. That's correct. But the plot_timeseries worked?

yarikoptic, 2020-04-21:
> Ok that's a recording extractor so it should not have units. That's correct. But the plot_timeseries worked, so it seems to be loaded fine

yeap, as could be seen on https://github.com/vandermeerlab/data-formats/blob/HEAD/yoh_try_spikeinterfaces.ipynb (sorry -- forgot to paste in this issue)

yarikoptic, 2020-04-21:
so any ideas on how to resolve/progress forward?

alejoe91, 2020-04-21:
I don't see the big here. From the notebooks the error seems to be a duplicate of https://github.com/SpikeInterface/spikeextractors/issues/369#issuecomment-617365173

alejoe91, 2020-04-21:
Just to be clear: plot_raster expects a SortingExtractor object, and here you are passing a RecordingExtractor

yarikoptic, 2020-04-22:
> Just to be clear: plot_raster expects a SortingExtractor object, and here you are passing a RecordingExtractor

d'oh -- right -- I see that now on https://spikeinterface.readthedocs.io/en/latest/getting_started/plot_getting_started.html .  Adding a check alike `if not isinstance(whateverpassed, SortingExtractor): raise ValueError("Please pass an instance of a SortingExtractor not %s" % whateverispassed)` would have prevented me tripping over it.  Feel welcome to close as is or with a suggested "fix".

yarikoptic, 2020-04-22:
another nice user-friendly enhancement could be to bind all those `plot_` functions to corresponding classes upon import of the `spikeinterface.widgets` so I could just use `.plot<TAB>` to complete for available plotting functions in a corresponding class

alejoe91, 2020-04-22:
> d'oh -- right -- I see that now on https://spikeinterface.readthedocs.io/en/latest/getting_started/plot_getting_started.html . Adding a check alike `if not isinstance(whateverpassed, [...]

I think this is a good idea for the `spikewidgets` package. We should add some assertions in general

alejoe91, 2020-04-22:
> another nice user-friendly enhancement could be to bind all those `plot_` functions to corresponding classes upon import of the `spikeinterface.widgets` so I could just use `.plot<TAB>` to complete for available plotting functions in a corresponding class

Maybe we could do it with a lazy load, but honestly I believe that the widget naming and docstring are quite explanatory :)

closed 2020-04-22 by alejoe91