SpikeInterface/spikeextractors #369: Neuralynx: ss.run_klusta() crashes on Neuralynx data due to broadcasting issue

yarikoptic, 2020-04-21:
NB sorry for the flood

```python
ss.run_klusta(recording)
```
leads to
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-10-cb71611d47a1> in <module>
----> 1 ss.run_klusta(recording)

~/anaconda-5.2.0-2.7/envs/mvdm-nept/lib/python3.7/site-packages/spikesorters/sorterlist.py in run_klusta(*args, **kwargs)
    195         The spike sorted data
    196     """
--> 197     return run_sorter('klusta', *args, **kwargs)
    198 
    199 

~/anaconda-5.2.0-2.7/envs/mvdm-nept/lib/python3.7/site-packages/spikesorters/sorterlist.py in run_sorter(sorter_name_or_class, recording, output_folder, delete_output_folder, grouping_property, parallel, verbose, raise_error, **params)
     79                          parallel=parallel, verbose=verbose, delete_output_folder=delete_output_folder)
     80     sorter.set_params(**params)
---> 81     sorter.run(raise_error=raise_error)
     82     sortingextractor = sorter.get_result()
     83 

~/anaconda-5.2.0-2.7/envs/mvdm-nept/lib/python3.7/site-packages/spikesorters/basesorter.py in run(self, raise_error)
    119     def run(self, raise_error=True):
    120         for i, recording in enumerate(self.recording_list):
--> 121             self._setup_recording(recording, self.output_folders[i])
    122 
    123         # dump again params because some sorter do a folder reset (tdc)

~/anaconda-5.2.0-2.7/envs/mvdm-nept/lib/python3.7/site-packages/spikesorters/klusta/klusta.py in _setup_recording(self, recording, output_folder)
     78             raw_filename = output_folder / 'recording.dat'
     79             dtype = 'int16'
---> 80             recording.write_to_binary_dat_format(raw_filename, time_axis=0, dtype=dtype, chunk_mb=500)
     81 
     82         if p['detect_sign'] < 0:

~/proj/misc/spikeextractors/spikeextractors/recordingextractor.py in write_to_binary_dat_format(self, save_path, time_axis, dtype, chunk_size, chunk_mb)
    715         '''
    716         write_to_binary_dat_format(self, save_path=save_path, time_axis=time_axis, dtype=dtype, chunk_size=chunk_size,
--> 717                                    chunk_mb=chunk_mb)
    718 
    719     def get_sub_extractors_by_property(self, property_name, return_property_list=False):

~/proj/misc/spikeextractors/spikeextractors/extraction_tools.py in write_to_binary_dat_format(recording, save_path, file_handle, time_axis, dtype, chunk_size, chunk_mb)
    301         chunk_size = int(chunk_size)
    302     elif chunk_mb is not None:
--> 303         n_bytes = recording.get_dtype().itemsize
    304         max_size = int(chunk_mb * 1e6)  # set Mb per chunk
    305         chunk_size = max_size // (recording.get_num_channels() * n_bytes)

~/proj/misc/spikeextractors/spikeextractors/recordingextractor.py in get_dtype(self)
    110 
    111     def get_dtype(self):
--> 112         return self.get_traces(channel_ids=[self.get_channel_ids()[0]], start_frame=0, end_frame=1).dtype
    113 
    114     def frame_to_time(self, frame):

~/proj/misc/spikeextractors/spikeextractors/extraction_tools.py in corrected_args(*args, **kwargs)
    660 
    661         # pass recording as arg and rest as kwargs
--> 662         get_traces_correct_arg = func(args[0], **kwargs)
    663 
    664         return get_traces_correct_arg

~/proj/misc/spikeextractors/spikeextractors/extractors/neoextractors/neobaseextractor.py in get_traces(self, channel_ids, start_frame, end_frame)
     85                                                                     channel_ids=channel_ids)
     86         # and then to uV
---> 87         scaled_traces *= self.additional_gain
     88 
     89         # fortunatly neo works with (samples, channels) strides

ValueError: non-broadcastable output operand with shape (1,1) doesn't match the broadcast shape (1,19)
```
full ipython notebook (with references to data source): https://github.com/vandermeerlab/data-formats/blob/HEAD/yoh_try_spikeinterfaces.ipynb

alejoe91, 2020-04-21:
I think I fixed that [today](https://github.com/SpikeInterface/spikeextractors/pull/366)! basically parsing the `dtype` string to a `np.dtype`. Can you pull the latest spikeextractors master?

yarikoptic, 2020-04-21:
Great, thanks @alejoe91 , ... I think I have done everything correctly to "enable" most recent changes (hard to say since `__version__` doesn't incorporate git describe), but seems to be the same.  updated that notebook with some versioning information I have tried to show to assure that have up to date spikeextractors

alejoe91, 2020-04-21:
Hi @yarikoptic, I pushed something like one hour ago, but the problem seems to be neo-related (sorry I went through the error trace quickly). I will need to ask @samuelgarcia..can you try to manually change the line that throws the error to:
`scaled_traces = np.array(scaled_traces) * self.additional_gain`?

Also, what is the output of `extractor.additional_gain`? Sorry, but I have little experience with the neo extractors (maybe for you it wouold be easier to start from something else, like an NWB or open ephys)

samuelgarcia, 2020-04-21:
The bug is in neo bridge I need to fix.
It is a broadcast problem for the gain.

samuelgarcia, 2020-04-21:
This bug is really strange.

Can you do
```
neuralynx_folder = 'MotivationalT-v2/R050/R050-2014-03-31_raw'
reader = neo.NeuralynxIO(dirname=neuralynx_folder)
print(reader.header['signal_channels']['units'])
print(reader.header['signal_channels']['units'].shape)

recording = se.NeuralynxRecordingExtractor(dirname=neuralynx_folder, seg_index=0)
print(additional_gain)
print(additional_gain.shape)
```

Or send me the file please ?

yarikoptic, 2020-04-21:
The entire directory (AKA folder) is available at http://datasets.datalad.org/?dir=/labs/mvdm/MotivationalT-v2/R050/R050-2014-03-31_raw since I am not sure which specific file is in question (optimal way to get the folder is via datalad or git + git-annex calls, but individual files could be fetched from the web as well).  Not sure if it would fit via email.  I guess I could also upload it somewhere else (e.g. https://gin.g-node.org/ - again optimally used with git and git-annex or datalad ;-))

output of adjusted
```python
neuralynx_folder = 'MotivationalT-v2/R050/R050-2014-03-31_raw'
reader = neo.NeuralynxIO(dirname=neuralynx_folder)
print(reader.header['signal_channels']['units'])
print(reader.header['signal_channels']['units'].shape)

recording = se.NeuralynxRecordingExtractor(dirname=neuralynx_folder, seg_index=0)
print(recording.additional_gain)
print(recording.additional_gain.shape)
```
is
```

['uV' 'uV' 'uV' 'uV' 'uV' 'uV' 'uV' 'uV' 'uV' 'uV' 'uV' 'uV' 'uV' 'uV'
 'uV' 'uV' 'uV' 'uV' 'uV']
(19,)
[[1. 1. 1. 1. 1. 1. 1. 1. 1. 1. 1. 1. 1. 1. 1. 1. 1. 1. 1.]]
(1, 19)
```

yarikoptic, 2020-04-21:
> Hi @yarikoptic, I pushed something like one hour ago, but the problem seems to be neo-related (sorry I went through the error trace quickly). I will need to ask @samuelgarcia..can you try to [...]

oh -- I should have "seen" this -- it seems to resolve it, the thing is running now and laptop is getting noisier ;-)

samuelgarcia, 2020-04-22:
In this output additional_gain.shape looks good.

There is a bug in NeoBaseRecordingExtractor when channel_ids is not None (all channel), additional_gain must be reduce.

I think it was the bug add at the begining.

I will try to fix it.

alejoe91, 2020-04-22:
Fixed here https://github.com/SpikeInterface/spikeextractors/pull/375

closed 2020-04-22 by alejoe91