SpikeInterface/spikeextractors #378: Q: how to aggregate/pass multiple Neuralynx blocks/segments to sorting?

yarikoptic, 2020-04-22:
As identified in #364, should extract a single block/segment.  But if recordings come from the same electrodes location, it makes sense to pass all the data into spike sorting.  How to achieve that?

alejoe91, 2020-04-22:
You can easily concatenate recordings in time using the [`MultiRecordingTimeExtractor`](https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/multirecordingtimeextractor.py)

It takes a list of RecordingExtractors as input and concatenate them in time. It is still a RecordingExtractor, so it can be spike sorted as it is

yarikoptic, 2020-04-22:
cool, thanks! I will give it a shot

closed 2020-04-22 by yarikoptic

samuelgarcia, 2020-04-23:
Note that MultiRecordingTimeExtractor is not totally ideal because it is a virtual concatenate of all segment which is not so good for filter border effect.
The cool thing is that you can do multi segment sorting with sorters that do not handle it natively.