SpikeInterface/spikeextractors #379: memmapped spike features and `get_unit_spike_features`

aliddell, 2020-04-22:
JRCLUST stores some raw and filtered spike traces in large binaries, formatted `n_samples x n_channels x n_spikes` (I guess you'd call them snippets). I'd like to be able to access these traces as unit spike features, but in the current framework it doesn't seem possible to do this without reading everything into memory. Here's how I was thinking of tackling this:

- For each unit, store either a boolean array or list of its indices in a dict.
- When someone calls `get_unit_spike_features` requesting either the raw or filtered traces, return the array of snippets.
- When someone calls `get_unit_spike_features` requesting something else, redirect them via a call to `super().get_unit_spike_features`.

I know that overriding a non-abstract method [is frowned upon](https://spikeinterface.readthedocs.io/en/latest/sorting_extractor.html#build-a-sortingextractor), but it would be such a waste to implement this somewhere else.

alejoe91, 2020-04-22:
Hi! You can actually do it. The extractors have a function called allocate_array that creates a memmap file of a given shape. You can then set the features with the memmap object. See how it's done in the spiketoolkit.postprocessing.get_unit_waveforms :) I'll tell you more tomorrow..getting late!

aliddell, 2020-04-23:
This looks like `allocate_array` induces a copy of the original file, which is what you probably want to do when you're overwriting, but the files I'm talking about already exist and I just want to open them up read-only and index into them. From what I can tell, each unit opens a separate file into which spike snippets are extracted. I'm also not sure which indices the `indexes` argument in `set_unit_spike_features` is supposed to represent.

alejoe91, 2020-05-06:
Hi @aliddell sorry for the delay in getting back on this. 

> This looks like `allocate_array` induces a copy of the original file, which is what you probably want to do when you're overwriting, but the files I'm talking about already exist and I just want to [...]

You could open the files externally and set them as features. This would keep their memmap nature:
```
memmap_features = np.memmap('path-to-file', shape, etc)
# this should have (num_units, num_spikes, num_channels)

for i, unit in enumerate(sorting.get_unit_ids():
    features = memmap_features[i]  # still a memmap object
    sorting.set_unit_spike_features(unit, 'your_feature_name', features)
```

This way the features are just a pointer to the memmap object. Does this make sense?

>I'm also not sure which indices the `indexes` argument in `set_unit_spike_features` is supposed to represent.

In the `spiketoolkit.posptocessing` module we have functions to extract waveforms, compute pc scores, etc. This can also be done on a subset of spikes. In that case, the `indexes` argument can be passed to indicate which spike indexes the features refer to. When this is done, say for features called `waveforms`, another feature called `waveforms_idxs` is created with the correspinding spike indexes.

aliddell, 2020-05-07:
```python
memmap_features = np.memmap('path-to-file', shape, etc)
# this should have (num_units, num_spikes, num_channels)

for i, unit in enumerate(sorting.get_unit_ids():
    features = memmap_features[i]  # still a memmap object
    sorting.set_unit_spike_features(unit, 'your_feature_name', features)
```

By `num_units` above do you mean `num_samples`? I guess not because of the line  `features = memmap_features[i]  # still a memmap object` (but then how many samples/channel does that yield?). The array is not organized by unit, but rather by the order in which each spike was detected. So if I want to extract spikes by unit, I'd need to use a boolean mask or an array of indices, either way it loads the data into memory. 😕

alejoe91, 2020-05-07:
Hi @aliddell 

I think I misread your initial message. In case you had a file with dimensions `(num_units, num_spikes, num_channels, num_samples)` (forgot the last dimension in my code snippet), then it would be possible, but I see that you have all the spikes for all units in a single channel. Sorry about the confusion!

In this case I think that what you suggest can be a viable solution. I think it's fine to override the `get_unit_spike_features` for a specific extractor, if it saves up a lot in performance. 
Note that the array of indexes can be saved as a propert, so you wouldn't need an extra internal dictionary (somehting like `jrc_waveform_idxs`).

aliddell, 2020-05-07:
OK. I have one or two bugfixes for my extractors anyway, I'll submit a PR.

closed 2020-05-11 by aliddell