SpikeInterface/spikeextractors #381: SortingExtractor object, change spike_train

llobetv, 2020-04-27:
Hi,
I try for several function (merge, remove redundant, alignement) to change spike_train value from sorting object but it appear that in each sorting object the original spike_train list is save with different name. 

What is the best way to have something stable? (switch to sort_cache everytimes?)

Cheer,
Victor

alejoe91, 2020-04-27:
Hi Victor,

Unfortunately as you noticed there is no proper way of changing the spiketrains outside of the curation module of spiketoolkit, and this is the case because we abstract the retrieval of the data with the abstract methods (e.g. `get_unit_spike_train`).

If you have suggestions (like removing duplicates) for some standard modifications they should be implemented in the curation module. For your own use, you could retrieve all the spike trains, modify them, make a new `NumpySortingExtractor` with the modified spike trains, and cache it.

Cheers
Alessio

llobetv, 2020-04-27:
OK, 
Then we'll probably try to implemented some common function in the curation module,

Thanks

closed 2020-04-27 by llobetv