SpikeInterface/spikeextractors #387: AssertionError in KiloSortRecordingExtractor when channel_map is a subset (active channels only) of np.arange(n_channels_dat)

aliddell, 2020-05-07:
I don't know when (or if) the `channel_map` format changed from when [this dataset](http://data.cortexlab.net/singlePhase3/) was sorted, but I run into this assertion error when I try to load the dataset with a `KiloSortRecordingExtractor`:

```python
basedir = "/path/to/singlePhase3/data/"
ex = se.extractors.kilosortextractors.KiloSortRecordingExtractor(basedir)
```

gives me 

```ipython
c:\users\alanc\source\repos\spikeinterface\spikeextractors\spikeextractors\extractors\kilosortextractors\kilosortextractors.py in __init__(self, folder_path)
     12
     13     def __init__(self, folder_path):
---> 14         PhyRecordingExtractor.__init__(self, folder_path)
     15
     16

c:\users\alanc\source\repos\spikeinterface\spikeextractors\spikeextractors\extractors\phyextractors\phyextractors.py in __init__(self, folder_path)
     27         elif (phy_folder / 'channel_map.npy').is_file():
     28             channel_map = list(np.squeeze(np.load(phy_folder / 'channel_map.npy')))
---> 29             assert len(channel_map) == self.params['n_channels_dat']
     30         else:
     31             channel_map = list(range(self.params['n_channels_dat']))

AssertionError:
```

Looking at the code it's not clear to me how the recordings handle inactive channels.

alejoe91, 2020-05-07:
Yep, the extractor assumes that there are no missing channels. Do you know how kilosort handle this? 

Maybe just removong that assertion and printing a warning message could work

aliddell, 2020-05-07:
> Do you know how kilosort handle this?

If I had to guess, I'd say Kilosort probably does what JRCLUST does and ignores samples from those channels when processing.

I tried replacing the assert with a warn, ran into another assert:

```
~\source\repos\SpikeInterface\spikeextractors\spikeextractors\extractors\bindatrecordingextractor\bindatrecordingextractor.py in __init__(self, file_path, sampling_frequency, numchan, dtype, recording_channels, time_axis, geom, offset, gain, is_filtered)
     36         if recording_channels is not None:
     37             assert len(recording_channels) == self._timeseries.shape[0], \
---> 38                'Provided recording channels have the wrong length'
     39             self._channels = recording_channels
     40         else:

AssertionError: Provided recording channels have the wrong length
```

alejoe91, 2020-05-08:
I see. So in that case we have to access the `self._timeseries` of the provided channels only

`self._timeseries[recording_channels]`

aliddell, 2020-05-08:
That's right, that's how I would (and do) access them (`self._timeseries` is `nchannels x nsamples`, correct?).

alejoe91, 2020-06-12:
Hi @aliddell 

Sorry for the delay here! Exactly the timeseries should have that shape (you can double check). 
Were you able to solve the issue? In that case, could you make a PR?

MarineChap, 2021-02-11:
Hi all, 
We ran into the same problem in our lab. 
```
Traceback (most recent call last):
  File "/home/kasia/.config/spyder-py3/temp.py", line 11, in <module>
    recording = PhyRecordingExtractor('/media/kasia/K/Neuropixeldata/LS_K_7/120920/Neuropixel_data/Processed/catgt_120920-VR_g0')
  File "/home/kasia/miniconda3/envs/fklab/lib/python3.7/site-packages/spikeextractors/extractors/phyextractors/phyextractors.py", line 29, in __init__
    assert len(channel_map) == self.params['n_channels_dat']
```
There is only 260 channels in channel_map.npy while in params there is 384. 
Did you find a solution for this?

Thanks

alejoe91, 2021-02-11:
Hi @MarineChap 

Yes! I was planning to make a PR by the end of the day actually (great timing :) )

Cheers
Alessio

alejoe91, 2021-02-11:
@MarineChap @aliddell 

Can you try this PR: https://github.com/SpikeInterface/spikeextractors/pull/570 ?

MarineChap, 2021-02-11:
Nice ! Testing it now

closed 2021-02-12 by alejoe91