SpikeInterface/spikeextractors #390: NwbRecordingExtractor.write_recording() error

alejoe91, 2020-05-13:
Hi @bendichter @luiztauffer 

When I try to export a recording to NWB using pynwb 1.3.0 and hmdf 1.6.1 I get the following error:

```
ValueError: row data keys don't match available columns you supplied 0 extra keys: set() and were missing 2 keys: {'rel_x', 'rel_y'}
```

The sorting extractor seems to work fine :)

alejoe91, 2020-05-13:
Here's the full traceback:
`se.NwbRecordingExtractor.write_recording(recording_cache, 'si_tutorial.nwb')`

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-41-ee5fceb2441c> in <module>
----> 1 se.NwbRecordingExtractor.write_recording(recording_cache, 'si_tutorial.nwb')

~/Documents/Codes/spike_sorting/spikeinterface/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py in write_recording(recording, save_path, metadata)
    555                 recording=recording,
    556                 nwbfile=nwbfile,
--> 557                 metadata=metadata
    558             )
    559 

~/Documents/Codes/spike_sorting/spikeinterface/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py in add_electrodes(recording, nwbfile, metadata)
    367                     location='unknown',
    368                     filtering='none',
--> 369                     group=grp,
    370                 )
    371         electrode_table = nwbfile.electrodes

~/anaconda3/envs/spikeinterface/lib/python3.6/site-packages/hdmf/utils.py in func_call(*args, **kwargs)
    481 
    482             if is_method:
--> 483                 return func(args[0], **parsed['args'])
    484             else:
    485                 return func(**parsed['args'])

~/anaconda3/envs/spikeinterface/lib/python3.6/site-packages/pynwb/file.py in add_electrode(self, **kwargs)
    532                 d.pop(col_name)  # remove args from d if not set
    533 
--> 534         call_docval_func(self.electrodes.add_row, d)
    535 
    536     @docval({'name': 'region', 'type': (slice, list, tuple), 'doc': 'the indices of the table'},

~/anaconda3/envs/spikeinterface/lib/python3.6/site-packages/hdmf/utils.py in call_docval_func(func, kwargs)
    362 def call_docval_func(func, kwargs):
    363     fargs, fkwargs = fmt_docval_args(func, kwargs)
--> 364     return func(*fargs, **fkwargs)
    365 
    366 

~/anaconda3/envs/spikeinterface/lib/python3.6/site-packages/hdmf/utils.py in func_call(*args, **kwargs)
    481 
    482             if is_method:
--> 483                 return func(args[0], **parsed['args'])
    484             else:
    485                 return func(**parsed['args'])

~/anaconda3/envs/spikeinterface/lib/python3.6/site-packages/hdmf/common/table.py in add_row(self, **kwargs)
    382                     'row data keys don\'t match available columns',
    383                     'you supplied {} extra keys: {}'.format(len(extra_columns), extra_columns),
--> 384                     'and were missing {} keys: {}'.format(len(missing_columns), missing_columns)
    385                 ])
    386             )

ValueError: row data keys don't match available columns
you supplied 0 extra keys: set()
and were missing 2 keys: {'rel_y', 'rel_x'}
```

The extractor has locations, so I'm not sure why it's failing.

alejoe91, 2020-05-13:
Here is a minimal reproducible example:

```
import spikeextractors as se

rec, sort = se.example_datasets.toy_example()
se.NwbRecordingExtractor.write_recording(rec, 'si_test.nwb')
```

alejoe91, 2020-05-13:
seems related to this: https://github.com/SpikeInterface/spikeextractors/issues/323

luiztauffer, 2020-05-15:
You're right, the problem is that [this bug](https://github.com/NeurodataWithoutBorders/pynwb/issues/1204) wasn't solved yet in `pynwb==1.3.0`
If you try if `pynwb==1.2.1` it should work. I tried with this and it worked:

```python
import spikeextractors as se
import pynwb

# Write
rec, sort = se.example_datasets.toy_example()
se.NwbRecordingExtractor.write_recording(rec, 'si_test.nwb')

# Read
with pynwb.NWBHDF5IO('si_test.nwb', 'r') as io:
    nwbfile = io.read()
    print(nwbfile)
```

alejoe91, 2020-05-15:
Thanks @luiztauffer 

However I found that with 1.2.1 groups and locations are not reloaded property. Can you diiuble check that on your side? 

The `rec` has locations, and you can add groups with `rec.set_channel_groups([0,0,1,1])`. When I load it back I have nan locations and all groups are 0.

luiztauffer, 2020-05-15:
I can reproduce the error also with `pynwb==1.3.0`
If you comment out the lines 364-365 to not pass the arguments rel_x and rel_y, the initial error does not happen. 

But the whole problem is we're mapping se location with pynwb rel_x and rel_y, and that will end up reproducing your error with the older version, which is: there's no rel_x/rel_y to be mapped to `_channel_properties[i]['location']` (line 196)

@bendichter we could map to `x` and `y` to have nwbextractor functional at least until https://github.com/NeurodataWithoutBorders/pynwb/issues/1204 gets solved? What you think?

luiztauffer, 2020-05-15:
I also found out that we are saving the location in the wrong place, now the correct place seems to be `self._key_properties['location']`? @alejoe91 
If that's the case, I can fix this part right now

alejoe91, 2020-05-15:
Oups I missed that! You can use self.set_channel_locations(location, channel_id). Same for groups (set_channel_groups). Probably that's why they where not loaded properly

alejoe91, 2020-05-15:
@luiztauffer I can quickly fix it, or are you on it already?

luiztauffer, 2020-05-15:
be my guest :)
Just make sure for now to use `pynwb=1.2.1`, which creates rel_x and rel_y in a way that won't  raise the initial error.

luiztauffer, 2020-05-15:
line 188 also might need a change?
`self._channel_properties[i]['gain'] = gains[ind]`

alejoe91, 2020-05-15:
I'll do it :) I forgot to change it. Properties should be set with the `set_channel_property`

bendichter, 2020-06-04:
update: [the real fix for this](https://github.com/NeurodataWithoutBorders/pynwb/pull/1246
) has been merged to dev on pynwb. Once it is released, we can change this code back to accepting the latest version of pynwb, and forbid the specific offending pynwb version.

alejoe91, 2020-06-04:
Awesome! I'll start opening a PR and install the dev version to see if it works :)

closed 2020-07-15 by alejoe91