SpikeInterface/spikeextractors #396: Request for alternative prb file format

lfrank, 2020-05-27:
I'm starting to work with SpikeInterface (and thanks to all of you for developing it) and I found the current prb file hard to work with. It's very close to, but not identical to, a json.dump of a dictionary, which is what I tried first to create it, but of course that didn't work.

As such I had to write more detailed code by hand. It would be great if there were alternative formats supported, including a simple json dump so that someone like me who is starting with a different structure for probes and electrodes (in this case an NWB file with a new probe extension based on a yaml file) can write to it more easily.

An ideal solution from my perspective would take advantage of the probe extension we've developed for nwb (see https://anaconda.org/NovelaKRK/ndx-fl-novela) and NWB ElectrodeGroups as the channel groups, but of course I don't know how many labs will eventually adopt our probe extension or a version thereof.

samuelgarcia, 2020-05-27:
@lfrank 
Cool. This is something I want since 6 month and I need someone to help for that.
I was not awre of this ndx-fl-novela.

My idea is to have a python module with some classes (Probe but not only) that would represent this problem.

And this problem is not only probe geometry, for a setup it is:
  * several probe geometry handling
  * wiring between probe and recording system
  * channel groups
  * maybe 3D position of probe
  * labelling

In my mind, this should absolutly not come from a file format representation BUT a set of classes.
And thoses classes could be dumped into a format (ndx-fl-novela, prb, csv, ..) that store partially (only group, or group + 2D geometry, or everything...).
I don't think we should choose NWB for this as it is a too big dependency just for probe handling.

I am not able to see this "ndx-fl-novela" on github could you link it ?

lfrank, 2020-05-27:
Hi Garcia. That all sounds great. I agree that all that information is required; we handle all that with mappings from the NWB electrode table to the probe information and make each ElectrodeGroup a probe.

I will say that I think there is real value in pushing toward NWB as the standard backend for this rather than coming up with yet another format, so if that’s something you’re willing to consider, that would be great. 

The novela extension is being developed in collaboration with a company that is volunteering their time, and they want to make sure everything is bullet proof before they release, which is why it isn’t there now. I’ll give them a nudge and see if I can get their repo made public.

Loren
--
Loren M. Frank

Professor  
Howard Hughes Medical Institute
Kavli Institute for Fundamental Neuroscience
Department of Physiology 
University of California, San Francisco
Office Phone: 415-502-7357

lfrank, 2020-06-03:
Just an update: the current version of our probe extension is available at https://github.com/NovelaNeuro/ndx-franklab-novela

samuelgarcia, 2020-11-30:
Hi @lfrank 
I have started a package to handle probe geometry and wiring to device.
(Also include some 2d or 3d plotting)

This is here:
https://probeinterface.readthedocs.io
https://github.com/SpikeInterface/probeinterface

The idea is NOT to make a new format but rather an API (like spikeextractor) that handle many formats (NWB, PRB, MEAREC ...)
I plan to handle the NWB geometry handling in this package.

Could you have a look at this package to make some feedback ?
The package is totally new (started last Tuesday), so feedback are welcome.

And maybe if you have time help me to make read/write to NWB format ?

lfrank, 2020-12-01:
Thanks for doing this; I think it’s a great idea. 

For the NWB format, our probe extension is in the ndx-franklab-novela extension (https://github.com/NovelaNeuro/ndx-franklab-novela <https://github.com/NovelaNeuro/ndx-franklab-novela>). And code that reads from a probe object into a DataJoint schema can be found in the Probe class within  the common/common_device.py file from https://github.com/LorenFrankLab/nwb_datajoint <https://github.com/LorenFrankLab/nwb_datajoint>. Finally, you can download a test NWB file which includes a probe from https://drive.google.com/file/d/1-0xdEn8oVT6lexBc8mlv7ls84oUvxU4u/view?usp=sharing <https://drive.google.com/file/d/1-0xdEn8oVT6lexBc8mlv7ls84oUvxU4u/view?usp=sharing>

Can you take a look at that and let me know what else you’d need to know to read our extension? 

As far as what you have so far, here are a few additional thoughts:

From my perspective it is more intuitive to have a probe correspond to a single physical device, which might have one or more shanks or some other organization of contacts (e.g. a net). I would therefore vote for renaming probe to either shank or perhaps channel group, and then having a probe consist of one or more shanks /  channel groups. I think that terminology is more consistent with the general usage in the field, where multi-shank probes are common.

I would also add that there is lot of other information that could be very useful in the long run in terms of keeping track of what sort of probe was used. This includes the following:

probe material (silicon, polyimide, SU-8, etc.)
electrode material (platinum, iridium, gold, etc.)
electrode plating (PEDOT, etc.)
probe thickness (I didn’t see this as something that could be added, but I may have missed it)

Finally, some probes have built in electronics (e.g. Neuropixels, the polymer probes we’re working on) so having a place for optional information about the electronics would be great.

Loren

samuelgarcia, 2020-12-02:
Hi @lfrank 
Thanks a lot for your feedback.

I will have a deeper look in this NWB format integration.

Would you be OK for public discussion on github issues here
https://github.com/SpikeInterface/probeinterface/issues

some of your feedback overlap some other feedback.