SpikeInterface/spikeextractors #398: Suggestions on OpenEphyRecordingExtractor

MingChingChiang, 2020-06-04:
Hi, 
First of all, thank you for the very helpful tutorial during the NWB user day remote workshop.
I’ve built my analysis pipeline based on the tutorial flow and it worked perfectly with short recordings (~10min long). However, when I attempt to load my 60min, 32 channel recording for analysis, not only the loading is slow but also the system memory exhausted easily, and my IDE just shutdown during the this step. I’ve tried to convert my recording to the NWB format but it didn’t work. 
I’m wondering is there any recommend solution to load or to preprocess my openephys data to improve the analysis procedure?
Thanks for the help!

alejoe91, 2020-06-04:
Hi @MingChingChiang 

Which open ephys format are you using? If it's the OpenEphys format (not the binary), then unfortunately the data are loaded in full in memory, because different channels are saved in different files. Is this your case? (for the future, I highly recommend using the binary or nwb formats).

For now, you could try to write a script to convert the Open Ephys format to a binary file, which is much more efficient!

samuelgarcia, 2020-06-04:
@alejoe91 with a wrapper on top on neo we could avoid the full memory loading. No ?

alejoe91, 2020-06-04:
@samuelgarcia maybe! We would need to keep several files open and retrieve chunks from the different files wheh `get_traces` is called

MingChingChiang, 2020-06-04:
> Hi @MingChingChiang
>
> Which open ephys format are you using? If it's the OpenEphys format (not the binary), then unfortunately the data are loaded in full in memory, because different channels are [...]
Thanks for the response! @alejoe91 
We did use Open ephys format instead of the binary or NWB. 

> For now, you could try to write a script to convert the Open Ephys format to a binary file, which is much more efficient!
And sure I will try to convert those files into binary format or NWB to speed up the analysis process.
Thanks for the advice!

closed 2020-06-19 by alejoe91