SpikeInterface/spikeextractors #399: NWB file open mode

lfrank, 2020-06-04:
Right now NWB files are opened in mode "a", which produces an error if the file is already open somewhere else. The code location is below:

`~/opt/anaconda3/envs/nwbdj/lib/python3.7/site-packages/spikeextractors/extractors/nwbextractors/nwbextractors.py in __init__(self, file_path, electrical_series_name)
    147         se.RecordingExtractor.__init__(self)
    148         self._path = file_path
--> 149         with NWBHDF5IO(self._path, 'a') as io:
    150             nwbfile = io.read()
    151             if electrical_series_name is not None:`

Can that be changed to 'r' so that the file can always be opened? This will be critical if multiple analyses are using the same data, which will be common.

When data need to be written to the NWB file, it could be reopened in mode 'a' or, even better, the .copy() method could be used to create a NWB file with links to the original, and data could be added to that. That is safer than adding data back to the original file.

alejoe91, 2020-06-04:
HI Loren,

See @magland PR: https://github.com/SpikeInterface/spikeextractors/pull/397

Just waiting for @bendichter and @luiztauffer to reply, as they have developed the NWBExtractors

bendichter, 2020-06-04:
thanks, @lfrank! I just approved @magland's fix, which should resolve your issue as well. Please let us know if you have any more problems. We would really like to continue to support this platform as a main highway for NWB conversion.

alejoe91, 2020-06-04:
https://github.com/SpikeInterface/spikeextractors/pull/397

closed 2020-06-04 by alejoe91