SpikeInterface/spikeextractors #400: Bug in NWB extractor - indeces for electrode table region are not channel_ids

lfrank, 2020-06-04:
In the code below the NWBExtractor uses the channel_ids to look up the indeces into the electrode table region of the e-series es. This produces an error for me where my electrode list has gaps (e.g. 1,2,3,5) but the electrodes.data[:] does not, as it indexes into the electrode table, which consists of consecutive entries.

The channel_ids here seem to refer to electrode_ids, which means that the code should be looking them up in nwbfile.electrodes to get their index in that table and then looking for that index in es.electrodes.data[:]

```
/Src/NWB/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py in get_traces(self, channel_ids, start_frame, end_frame)
    257             nwbfile = io.read()
    258             es = nwbfile.acquisition[self._electrical_series_name]
--> 259             table_ids = [list(es.electrodes.data[:]).index(id) for id in channel_ids]
    260             if np.array(channel_ids).size > 1 and np.any(np.diff(channel_ids) < 0):
    261                 sorted_idx = np.argsort(table_ids)
```

alejoe91, 2020-06-04:
Hi Loren,

Again asking for @bendichter  and @luiztauffer opinion on this :)

bendichter, 2020-06-04:
Thanks @lfrank. You are correct. We'll make the change

PR #401 (2020-06-05): fix get traces by index
fix #400
closed 2020-06-05 by alejoe91