SpikeInterface/spikeextractors #411: NeuroscopeSpikeExtractor

CodyCBakerPhD, 2020-06-18:
Hello, I was attempting to use the NeuroscopeSpikeExtractor function to retrieve spike times from .res and .clu files for use in the widgets and further processing. I noticed a repetitive error where one of the units, typically the last index, would be non-spiking (spike times array is empty) which causes issues with things as simple as plot_raster.

After digging into it, it seems the issue is with the NeuroscopeSpikeExtractor assuming that the cluster IDs start at 1 and go up to the number of units reported in the first line of the .clu file... However, the actual Neuroscope data suggests it is instead ordered from 0, ..., number of units - 1.

You can find a sample notebook illustrating the issue here: https://github.com/catalystneuro/buzsaki-lab-to-nwb/blob/refactor_v1.0.1/buzsaki_lab_to_nwb/original-code/Replicating_NeuroscopeSortingExtractor_Issue.ipynb

alejoe91, 2020-06-19:
HI @CodyCBakerPhD 

thanks for the report. Actually @alexmorley implemented the NeuroScopeExtractor.

We added now a new [base function](https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/sortingextractor.py#L703) for the SortingExtractor that can return unsorted spikes. This could return the cluster 0. 

For cluster 1, we could add an option`keep_mua_units` False by default, that would allow the user to include the mua clusters or not.

How does that sound? If you are getting familiar with the framework, would you like to make these changes in a PR? 

Cheers
Alessio

CodyCBakerPhD, 2020-06-19:
That sounds like the best option, it should be a pretty easy change to make. Just to make sure I have the process correct, I should make a new branch, commit the changes there, and await comments/testing on your end until you merge it into the master?

alejoe91, 2020-06-19:
You should fork the repo, make a new branch, make the changes, and then make a PR to spikeextractors/master from your fork :)

alexmorley, 2020-06-19:
Happy to review. You are right I don't think I considered `0` units in my implementation because we don't use them.

closed 2020-06-24 by alejoe91