SpikeInterface/spikeextractors #414: Help with using neuralynxextractor

ohnopleasehelpme, 2020-06-21:
Hello - I am using spikeinterface. So far, I just want to import my code into "recording" variable.

import spikeinterface.extractors as se
import matplotlib.pylab as plt
import numpy as np

directory = '~/PycharmProjects/SpikeGUI/data/'
recording = se.NeuralynxRecordingExtractor('~/PycharmProjects/SpikeGUI/data/')

When I do, I get the following error:

Backend MacOSX is interactive backend. Turning interactive mode on.
Traceback (most recent call last):
  File "~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-6a042b03c901>", line 1, in <module>
    runfile('~/PycharmProjects/SpikeGUI/nueralynx_extract.py', wdir='~/PycharmProjects/SpikeGUI')
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "~/PycharmProjects/SpikeGUI/nueralynx_extract.py", line 11, in <module>
    recording = se.NeuralynxRecordingExtractor('~/PycharmProjects/SpikeGUI/data/')
TypeError: __init__() takes 1 positional argument but 2 were given

I am wondering why am I getting this error and how do I input the directory into "NeuralynxRecordingExtractor". Thank you.

alejoe91, 2020-06-21:
@samuelgarcia missing seg_index and block_index?

samuelgarcia, 2020-06-22:
Hi,
the syntax is 
`rec = se.NeuralynxRecordingExtractor(dirname='/path/to/my/neuralynx/folder')`
and if you have a multi segment file you must specify which segment youwant to read
`rec = se.NeuralynxRecordingExtractor(dirname='/path/to/neuralynx_folder', seg_index=0)`

note that there is no args bu only kwargs mapped to neo rawio.
If you want to known how many segments:
```
reader = neo.rawio.NeuralynxRawIO(dirname='/path/to/my/neuralynx/folder')
print(reader)
```

ohnopleasehelpme, 2020-06-22:
I have the neuralynx dataset in that folder as a .ncs filetype, for example, "CSC1.ncs" too - do I have to convert it into a different format?

samuelgarcia, 2020-06-22:
No. ncs is the natural neuralynx format.
But neuralynx is not one only file but a bunch of file regrouped in a folder.
You need to provide the folder that contain thoses files with  `dirname=...`

ohnopleasehelpme, 2020-06-22:
I see. Is there a guide somewhere (through spikeinterface, tutorials etc) where all the input variables are explained? 
For example, what does seg_index do? I set seg_index=0 for now but when i change it to a larger number, I see more of the trace.

alejoe91, 2020-06-22:
The NeuraLynxExtractors use `neo` as backend. There's plenty of documentation on the concepts of segments and blocks here: http://neuralensemble.org/neo/

ohnopleasehelpme, 2020-06-25:
So another issue - when I add the following sorter variable to the se.NerualynxRecordingExtractor function:

rec, sorting = se.NeuralynxRecordingExtractor(dirname='~/PycharmProjects/SpikeGUI/data', seg_index=10)

I get the error:

Traceback (most recent call last):
  File "~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-6a042b03c901>", line 1, in <module>
    runfile('~/PycharmProjects/SpikeGUI/nueralynx_extract.py', wdir='~/PycharmProjects/SpikeGUI')
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "~/PycharmProjects/SpikeGUI/nueralynx_extract.py", line 9, in <module>
    rec, sorting = se.NeuralynxRecordingExtractor(dirname='~/PycharmProjects/SpikeGUI/data', seg_index=10)
TypeError: cannot unpack non-iterable NeuralynxRecordingExtractor object

I was following the tutorial in which the toyExample could output spiking data for a raster plot. Is this not the way for extracting neuralynx data? Thank you again!

alejoe91, 2020-06-25:
Hi,

The RecordingExtractor (like the one you use) only returns a recording, not a sorting. The toy example is a function that generates both a recording and a sorting, so it's something else :)

You should use:
`rec = se.NeuralynxRecordingExtractor(dirname='~/PycharmProjects/SpikeGUI/data', seg_index=10)
`

ohnopleasehelpme, 2020-06-25:
So I just have to use the 
rec = se.NeuralynxRecordingExtractor(dirname='~/PycharmProjects/SpikeGUI/data', seg_index=10)

to extract and sorting function to sort?

alejoe91, 2020-06-25:
Exactly. You load the recording and then sort it to obtain the sorting. In fact I guess your dataset doesn't have spike sorted information. Check out the tutorial for more information: https://spikeinterface.readthedocs.io/en/latest/getting_started/plot_getting_started.html
You can see that the `sorting_true` is not used (until the end).

samuelgarcia, 2020-06-25:
I think you want:
`sorting = se.NeuralynxSortingExtractor(dirname='~/PycharmProjects/SpikeGUI/data', seg_index=10)`

ohnopleasehelpme, 2020-06-25:
Ahh ok, thank you again!! It worked :)

closed 2020-06-25 by alejoe91

ohnopleasehelpme, 2020-06-25:
The sorter works, but the raster plot does not: 

sorted = se.NeuralynxSortingExtractor(dirname='~/PycharmProjects/SpikeGUI/data', seg_index=0)
w_rs = sw.plot_rasters(sorted)

ERROR:

 self.ax.set_xlim(visible_start_frame, visible_end_frame)
Exception ignored in: <function SortingExtractor.__del__ at 0x10b4b60e0>
Traceback (most recent call last):
  File "~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/spikeextractors/sortingextractor.py", line 28, in __del__
    if self._tmp_folder is not None:
AttributeError: 'NeuralynxSortingExtractor' object has no attribute '_tmp_folder'

ohnopleasehelpme, 2020-06-25:
And it opens a blank plot for the raster plot

alejoe91, 2020-06-26:
Do you have units? 

`print(sorted.get_unit_ids())`