SpikeInterface/spikeextractors #419: A little more clarification with Neuralynx SpikeInterface

ohnopleasehelpme, 2020-06-26:
**Here is my code:**
import spikeinterface.extractors as se
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw
import matplotlib.pylab as plt
import numpy as np

rec = se.NeuralynxRecordingExtractor(dirname='~/PycharmProjects/SpikeGUI/data', seg_index=0)
spikes = se.NeuralynxSortingExtractor(dirname='~/PycharmProjects/SpikeGUI/data', seg_index=0)

#set variables of channel_ids, sampling frequency and the number of channels
channel_ids = rec.get_channel_ids()
fs = rec.get_sampling_frequency()
num_chan = rec.get_num_channels()

# Print the variables
print('Channel ids:', channel_ids)
print('Sampling frequency:', fs)
print('Number of channels:', num_chan)

# recording probe data
recording_prb = rec.load_probe_file('custom_probe_3e.prb')

# plot the time-series
w_ts = sw.plot_timeseries(rec, trange=[0, 5])

#plot the raster plot
w_rs= sw.plot_rasters(spikes, trange=[0, 5])

# bandpass the data
rec_f = st.preprocessing.bandpass_filter(rec, freq_min=300, freq_max=6000)

# plot the filtered data
f_ts = sw.plot_timeseries(rec_f)

 **OUTPUT:**
runfile('~/PycharmProjects/SpikeGUI/nueralynx_extract.py', wdir='~/PycharmProjects/SpikeGUI')
~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/elephant/pandas_bridge.py:22: DeprecationWarning: pandas_bridge module will be removed in Elephant v0.8.x
  DeprecationWarning)
17:54:52 [I] klustakwik KlustaKwik2 version 0.2.6
Backend MacOSX is interactive backend. Turning interactive mode on.
Channel ids: [4, 6, 7]
Sampling frequency: 32556.0
Number of channels: 3
'end_time' set to 1024
~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/spikewidgets/widgets/rasterswidget/rasterswidget.py:91: UserWarning: Attempting to set identical left == right == 0.0 results in singular transformations; automatically expanding.
  self.ax.set_xlim(visible_start_frame, visible_end_frame)
WARNING! No channel location given. Add dummy location.
Warning! The recording is already filtered, but Mountainsort4 filter is enabled. You can disable filters by setting 'filter' parameter to False
'end_time' set to 1024

**Issues:**
(1) The units on the axis should go from 0-5 seconds, but instead shows 0-0.028 (s) instead

[image]

(2) Raster plot does not work still... 
[image]

Is there a site that includes how to add the inputs into functions? Sometimes the input is different than the online tutorial.

alejoe91, 2020-06-26:
You don't have the spikes variable, so it's no supposed to work. You only have the rec object.

The raster plot will work after running spikesprting. You can check out this video tutorial as well: https://youtu.be/nWJGwFB7oII

You can see the inputs/outputs for all funcrions here: https://spikeinterface.readthedocs.io/en/latest/api.html

alejoe91, 2020-06-26:
@ohnopleasehelpme sorry I didn't read the code well. Can you you use the `code` style? 

Do you actually have units? what is the output of: `print(len(spikes.get_unit_ids()))`

ohnopleasehelpme, 2020-06-26:
Sorry about the bold font! I thought it would look different so no worries. 

Apparently the length of the spike variable is 0 - but when I run the spike detection algorithm on matlab, for this data set I have about 47 spikes in the ~10 second sample trace. 

Also, the time axis seems to be off, I guess I am confused about how the seg_index should be used (or what other input variable could be used for running the "NeuralynxRecordingExtractor" function such as end time)

ohnopleasehelpme, 2020-06-26:
And btw, the video is really well done and really helpful! 

The jupyter notebook worked really well for me when I tried it a month ago. My source of confusion is stemming from what input variables are needed for the functions such as se.NeuralynxRecordingExtractor(input_1, input_2,.....input_n) or se.NeuralynxSortingExtractor(input_1, input_2,.....input_n)

alejoe91, 2020-06-26:
@ohnopleasehelpme (btw really nice github name eheheh)

So I think that your Neuralynx data are NOT spike sorted, so the `NeuralynxSortingExtractor` has no units in it. 

You should use spikeinterface to spike sort your data, so for now forget about the `NeuralynxSortingExtractor` and focus on the recording.

The `NeuralynxRecordingExtractor` take as input the `seg_index`, which is the segment number. This means that you recording has multiple segments (i.e. multiple recordings). The `seg_index` tell which of these segments you are loading, and you can load one at a time.

If you want to load, concatenate, and spike sort more than one segments, you can use a `MultiRecordingTimeExtractor`. You first need to figure out how many segments you have (let's call it `nseg`). Then you can do:

```
recording_list = []
dirname = '/PycharmProjects/SpikeGUI/data'
for seg_idx in range(nseg):
    recording_single = se.NeuralynxRecordingExtractor(dirname=dirname, seg_index=seg_idx)
    recording_list.append(recording_single)

recording = se.MultiRecordingTimeExtractor(recording_list)

# then you can go on with the preprocessing/spikesorting like in the tutorial/video
recording_f = st.preprocessing.bandpass_filter(recording)
sorting = ss.run_klusta(recording_f)

print(len(sorting.get_unit_ids()))
```

Hope this is clearer

ohnopleasehelpme, 2020-06-26:
Ok sounds good, I will try the suggestions and let you know how it goes. 

So as far as I understand, if I have a sample with 10000 points, then I have to concatenate multiple segments into a long, continuous trace? If my data set is from a single, continuous, 30 minute recording, do I still need to segment the trace?

How do I figure out nseg? Maybe using print(len(recording))?

Also, if the data isnt spike-sorted what function would I use to find the spikes? I thought it was the spike extractor?

Thank you again!

ohnopleasehelpme, 2020-06-26:
Probably would have been more professional to use my name like everyone else :)

alejoe91, 2020-06-26:
I think you can find the number of segments using neo (@samuelgarcia). If you have a single long file you don't need the concatenation.

The sorting extractor does not perform spike sorting, but it loads spike sorted data. For spike sorting you can use the `run_sorter` function, as shown in the tutorials.

ohnopleasehelpme, 2020-06-27:
For the online tutorial, the extractor only needed the data set location and then worked without seg_index which can then be plotted easily:... 

recording_folder = 'open-ephys-dataset/'
recording = se.OpenEphysRecordingExtractor(recording_folder)
w_ts = sw.plot_timeseries(recording, trange=[5, 10])

This is really nice and straightforward for openephys! What are there differences between neuralynx and openephys - the tutorial/openephys makes sense, but for neuralynx I just don't feel like I have a good understanding of the seg_index and why I am unable to plot just the trace with respect to trange=[]...

It would be really nice to just use the directory name, without having to set the seg_index.
rec = se.NeuralynxRecordingExtractor(dirname) 
w_ts = sw.plot_timeseries(rec, trange=[0, 10])

ohnopleasehelpme, 2020-06-27:
Yes! How do I just extract a single long continuous trace instead of breaking it down segment by segment and then putting it back together? Using se.NeuralynxRecordingExtractor

ohnopleasehelpme, 2020-06-28:
Update: this is what I have for the spikeGUI project. I am able to get the output folder for MS4 and phy:

import spikeinterface.extractors as se
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw
import matplotlib.pylab as plt
import numpy as np
import neo

reader = neo.NeuralynxIO(dirname='~/PycharmProjects/SpikeGUI/data')
print(reader)
print(reader.segment_count(0))

rec_list = [se.NeuralynxRecordingExtractor(dirname='~/PycharmProjects/SpikeGUI/data', seg_index=i)
            for i in range(reader.segment_count(0))]
rec = se.MultiRecordingTimeExtractor(rec_list)
a = range(reader.segment_count(0))

print(type(rec))
channel_ids = rec.get_channel_ids()
fs = rec.get_sampling_frequency()
num_chan = rec.get_num_channels()

print('Channel ids:', channel_ids)
print('Sampling frequency:', fs)
print('Number of channels:', num_chan)

w_ts = sw.plot_timeseries(rec, trange=[0, 10])

trace_snippet = rec.get_traces(start_frame=int(fs*0), end_frame=int(fs*2))
print('Traces shape:', trace_snippet.shape)

recording_prb = rec.load_probe_file('custom_probe.prb')
print('Channels after loading the probe file:', recording_prb.get_channel_ids())
print('Channel groups after loading the probe file:', recording_prb.get_channel_groups())

w_elec = sw.plot_electrode_geometry(recording_prb, markersize=50)

print(ss.installed_sorters())
print(ss.get_default_params('mountainsort4'))

rec_f = st.preprocessing.bandpass_filter(recording_prb, freq_min=300, freq_max=6000)
w = sw.plot_timeseries(rec_f, trange=[0, 10])

sorting_MS4 = ss.run_mountainsort4(rec_f, adjacency_radius=50)
#sorting_KL = ss.run_klusta(recording=rec_f)

st.postprocessing.export_to_phy(rec_f, sorting_MS4, output_folder='phy')

(1) How do I know that the spikes were sorted? 
print(len(sorting_MS4.get_unit_ids()))=1

(2) how do I plot the sorted spikes on Phy? When I run Phy, I get the following error (after following the guide: https://github.com/cortex-lab/phy)

MacBook-Pro:~/PycharmProjects/SpikeGUI/phy
$ phy template-gui params.py
15:33:45.843 [E] __init__:62          An error has occurred (RuntimeError): ndarray subclass __array_wrap__ method returned an object which was not an instance of an ndarray subclass
Traceback (most recent call last):
  File "~/opt/miniconda3/envs/phy2/bin/phy", line 8, in <module>
    sys.exit(phycli())
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/gui.py", line 198, in template_gui
    controller = TemplateController(model=load_model(params_path), dir_path=dir_path, **kwargs)
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 1178, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 297, in __init__
    self._load_data()
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 400, in _load_data
    self.sparse_features = self._load_features()
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 710, in _load_features
    cols = self._read_array(self._find_path('pc_feature_ind.npy'), mmap_mode='r')
  File "~/opt/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 426, in _read_array
    return read_array(path, mmap_mode=mmap_mode).squeeze()
RuntimeError: ndarray subclass __array_wrap__ method returned an object which was not an instance of an ndarray subclass

(phy2)

alejoe91, 2020-06-28:
The recording is sorted and you only got one unit from ms4. Phy needs at least 2 units, that's why it's throwing the error. You can try another sorter! 
Everything seems to be running fine. Let me know if you have other issues or we can close this.

Ps. Can you use the code block when you paste code?

ohnopleasehelpme, 2020-06-28:
Using the insert code on the github widgets?

closed 2020-07-03 by alejoe91