SpikeInterface/spikeextractors #422: Cachesortingextractor, tmp folder

llobetv, 2020-07-08:
Hi guys, 
I encounter an issue link to Cachesortingextractor, I have a ridiculous small tmp directroy so it is often times full (particulary when I launch a lot of analysis...) 
The issue is that Cachesortingextractor tmp directory use the default one, would it be possible to add a params or use tmp directory of the sorting object?

I you agree I make the modification myself.
Cheers,
Victor

samuelgarcia, 2020-07-08:
Hi.
There is already a params `save_path` to control this in Cachesortingextractor
https://github.com/SpikeInterface/spikeextractors/blob/f0d2653f6e55788150d0fcba4242aee5c75b8988/spikeextractors/cacheextractors.py#L103

maybe your problem is that you call a function that do not offer this save_path, no ?

llobetv, 2020-07-08:
Oh you are right I completly miss it... 
Thanks!

closed 2020-07-08 by llobetv

alejoe91, 2020-07-08:
The problem could also be that the tmp folder gets full for other reasons (e.g. waveforms).

Maybe we could add a spikeinterface function to change the TMP folder. On linux, you can do that by setting the `TMPDIR` variable.

From the terminal:
`export TMPDIR='path-to-new-tmp-dir'`

samuelgarcia, 2020-07-08:
Yes. Having a global tmp folder that could be set is a good ideaa.

llobetv, 2020-07-08:
Isn't what sorting.set_tmp_folder() method already do?

alejoe91, 2020-07-08:
That does it for one object only, but I think it would be nice to have the possibility to set the environment variable