SpikeInterface/spikeextractors #426: MemoryError: Unable to allocate 257. GiB for an array with shape (383, 360105397) and data type int16`

Jyh-Jang, 2020-07-13:
Dear,

I am reporting a problem of dealing with big data (257 GB), see below. Is there any alternative way to load this data with physical memory of 125 GB?

Thank you, 
JJ

`Traceback (most recent call last):
  File "/export/lib/from_external_packages/bin/spike_interface_pipeline.py", line 235, in <module>
    recording = run_extractor()
  File "/export/lib/from_external_packages/bin/spike_interface_pipeline.py", line 120, in run_extractor
    recording.write_recording(subrecording, OUTDIR / 'recording')
  File "/export/opt/anaconda2/envs/spikeinterface/lib/python3.7/site-packages/spikeextractors/extractors/spikeglxrecordingextractor/spikeglxrecordingextractor.py", line 88, in write_recording
    np.transpose(np.array(recording.get_traces(), dtype=dtype)).tofile(f)
  File "/export/opt/anaconda2/envs/spikeinterface/lib/python3.7/site-packages/spikeextractors/extraction_tools.py", line 625, in corrected_args
    get_traces_correct_arg = func(args[0], **kwargs)
  File "/export/opt/anaconda2/envs/spikeinterface/lib/python3.7/site-packages/spikeextractors/subrecordingextractor.py", line 40, in get_traces
    return self._parent_recording.get_traces(channel_ids=original_ch_ids, start_frame=sf, end_frame=ef)
  File "/export/opt/anaconda2/envs/spikeinterface/lib/python3.7/site-packages/spikeextractors/extraction_tools.py", line 625, in corrected_args
    get_traces_correct_arg = func(args[0], **kwargs)
  File "/export/opt/anaconda2/envs/spikeinterface/lib/python3.7/site-packages/spikeextractors/extractors/spikeglxrecordingextractor/**spikeglxrecordingextractor.py**", line 78, in get_traces
    recordings = self._timeseries[channel_idxs, start_frame:end_frame]
  File "/export/opt/anaconda2/envs/spikeinterface/lib/python3.7/site-packages/numpy/core/memmap.py", line 331, in __getitem__
    res = super(memmap, self).__getitem__(index)
**MemoryError: Unable to allocate 257. GiB for an array with shape (383, 360105397) and data type int16**`

alejoe91, 2020-07-13:
Hi @Jyh-Jang 

Can you paste the code that you are using?

Jyh-Jang, 2020-07-13:
Hi Alessio,

spike_interface_pipeline.py --binary_input_file /my/path1/18062020_alternationtask_day1_2_g0_t0.imec0.ap.bin --json_file /my/path2/Eric_day1_g0_imec0.json --out_dir /my/path3/ --sorter_name kilosort2 --extractor_type glx --pitch_for_glx 16 20 --use_preprocessing False --use_phy_export True

Tks, JJ

alejoe91, 2020-07-13:
Hi JJ,

What is the content of spike_interface_pipeline.py?

mhhennig, 2020-07-13:
Is there perhaps not enough disk space?

alejoe91, 2020-07-13:
Yes but it shouldn't be a problem as long as not all traces are retrieved at once!

mhhennig, 2020-07-13:
The problem occurs when the memmap array is created. Only reason I can see id that the disk is full.

gmaggi, 2020-07-14:
Hi @alejoe91 , I helped with the installation of spikeinterface in a machine that @Jyh-Jang is using. 
The content of spike_interface_pipeline.py is:
1.-

```
recording = se.SpikeGLXRecordingExtractor(args.INPUTFILE,
                                                    x_pitch = args.PITCHGLX[0],
                                                    y_pitch = args.PITCHGLX[1])
                                                            
subrecording = recording.load_probe_file(prb_file)
recording.write_recording(subrecording, OUTDIR / 'recording')
```

2.- 

```
ss.Kilosort2Sorter.set_kilosort2_path('/export/lib/KiloSort2/')
outSorter = ss.run_kilosort2(recording = InRecording,
                                       output_folder = pathToFolder,
                                       **params)
```

The params are (@Jyh-Jang correct them please if those are wrong):

```
"detect_threshold": 5,
"projection_threshold": [10, 4],
"preclust_threshold": 8,
"car": false,
"minFR": 0.1,
"minfr_goodchannels":0,
"freq_min": 150,
"sigmaMask": 30,
"nPCs":3,
"ntbuff":64,
"nfilt_factor":12

```

3.-

JJ also uses `st.postprocessing.export_to_phy` , but it seems that crash happens before that.

gmaggi, 2020-07-14:
The problem seems to be because an array of shape `(383, 360105397) and data type int16` (as posted in the error above by @Jyh-Jang ) can not be allocated. This would be expected as the `free` command shows the following:
```
             total        used        free      shared  buff/cache   available
Mem:           125G        1.0G        5.1G         15M        119G        124G
Swap:          128G        856M        127G

```

Thus, it seems there is no space in memory. is it possible to split the array in chunks, or analyze a subset of data using the current spikeinterface?

alejoe91, 2020-07-14:
Hi @gmaggi 

Indeed the KS2 internally makes a binary copy of the data in chunks. However, when you call the `recording.write_recording()` it uses the SpikeGLX one, that has no chunking. Try to run:

```
recording = se.SpikeGLXRecordingExtractor(args.INPUTFILE,
                                                    x_pitch = args.PITCHGLX[0],
                                                    y_pitch = args.PITCHGLX[1])
                                                            
subrecording = recording.load_probe_file(prb_file)

ss.Kilosort2Sorter.set_kilosort2_path('/export/lib/KiloSort2/')
outSorter = ss.run_kilosort2(recording = subrecording,
                                       output_folder = pathToFolder,
                                       **params)
```

You should not run into memory errors this way because chunking is applied internally :)

gmaggi, 2020-07-14:
Thanks Alejo!, we'll try and let you know.

gmaggi, 2020-07-16:
@alejoe91 , that works. Thanks again!. Feel free to close this ticket.

closed 2020-07-16 by alejoe91