SpikeInterface/spikeextractors #433: NWB file error

lfrank, 2020-08-07:
I had thought this was fixed, but version 0.8.4 from a pip install still uses 'a' rather than 'r' to open NWB files, which causes problems if the file is open somewhere else.

Would it be possible to get a new release with that fixed?

alejoe91, 2020-08-10:
Hi @lfrank 

Thanks for bringing this up. I guess this is because some properties (e.g. quality metrics) could be added by SpikeInterface. Maybe we could make 'r' default and optionally have 'a'. @bendichter what do you think?

bendichter, 2020-08-10:
@lfrank thanks for using the package and raising issues! The most recent version of spikeextractors is [0.8.4, released on May 13](https://pypi.org/project/spikeextractors/). It looks like there was one place where an 'a' should have been an 'r', and [Jeremy Magland fixed this on June 3](https://github.com/SpikeInterface/spikeextractors/commit/2573913157fd47459ff558f553b0bac8b2a45ec4). We just haven't cut an release since then. @alejoe91 what's the policy on cutting releases?

alejoe91, 2020-08-11:
I'll release all the packages including SpikeInterface this week!

bendichter, 2020-08-29:
@alejoe91 close?

closed 2020-08-29 by alejoe91