SpikeInterface/spikeextractors #438: Chunked File writing

alexmorley, 2020-08-20:
Suggestion: when writing / copying a file in chunks the filename could be `.{filename}.part` (and then renamed at the end) so that it's not possible to get a partially written file?

alejoe91, 2020-08-24:
Hi @alexmorley 

That's a good suggestion, thanks! 

However, this might be problematic for some sorters that expect a single binary file to be processed (e.g. klusta, kilosort2). We need to investigate whether they support this chunked format. Alternatively, we could have an option to write to a single or multiple files.

Cheers
Alessio

alexmorley, 2020-08-24:
Sorry, badly phrased suggest. I still mean to write to a single file. Only that while the file is not yet complete it should have .part appended so that if there is an uncaught error you can't mistake the file for a complete one.

alejoe91, 2020-08-24:
Ahhh! got it! ok that sounds good :)

Would you have time to make a PR? Otherwise I should be able to get to it over the next couple of weeks!