SpikeInterface/spikeextractors #446: Weird chunking issue

jthermiz, 2020-09-01:
I have a weird, elusive issue that I think is related to chunking for mda files. 

I'm able to run my spike sorting script fine with 5 seconds worth of data, but when I extend it to past 10 minutes it breaks. For some reason, I only get this issue on a particular machine. That indicates an environment issue, so I installed a conda environment from a working one, but it still breaks. I doubt this is an issue with the package, but I was hoping to get any advice / help on how to resolve this issue.

Error:
_ValueError: cannot reshape array of size 0 into shape (16,500)_
[weird_chunking_error.txt](https://github.com/SpikeInterface/spikeextractors/files/5154062/weird_chunking_error.txt)

Script:
[error_generating_script_py.txt](https://github.com/SpikeInterface/spikeextractors/files/5154067/error_generating_script_py.txt)

My environment:
[my_env.txt](https://github.com/SpikeInterface/spikeextractors/files/5154042/my_env.txt)

alejoe91, 2020-09-01:
Hi @jthermiz 

I think I have seen this error before and the reason is that in order to use whiten you need filtering as well. I will add an assertion to check that! Can you test with whiten and filter True, or with withen=False. 

A comment about the parallel arg. This is used when you have multiple groups to launch parallel jobs for each group (e.g. tetrodes). The num_workers arg is specific to ms4 and it is used to parallelize processing in the input.

Let me know if it solves the issue!
Alessio

jthermiz, 2020-09-01:
Hi @alejoe91 

Thanks for the quick response. Unfortunately, I tried all combinations of True / False for whiten and filter parameters and they all gave some version of the reshape error. 

Thanks for that explanation!

-John

alejoe91, 2020-09-01:
@jthermiz 

I tried your code with simulated data saved to MDA of the same size (16 channels, over 700 s + sub extractor with first 700 seconds) and it ran with no issues. 

Can you `pip install --upgrade spikeextractors spiketoolkit spikesorters spikecomparison spikeidgets`?

We just made a new release! https://github.com/SpikeInterface/spikeinterface/releases/tag/0.10.0

One last comment on the parameter logging. When you run a sorter, the a serialized version of the recording input and the spike sorter params are dumped automatically in the folder `outputfolder/spikeinterface_params.json`. So you don't need to dump them twice!

Let me know if the error persists with the latest versions.

Alessio

jthermiz, 2020-09-24:
@alejoe91 

Sorry for the delay in responding. I figured out the issue. The MDA file I tried to load was bad. I downloaded it from [Mendeley](https://data.mendeley.com/datasets/j2mfvnsz2t/1). The MDA should be 7GB, but Download All only downloads 448MB worth of data. When I just download the MDA, it correctly downloads all of the 7GB. When I pass that through, spikextractors things work as they should. So there's no issue with spikextractors!

-John

closed 2020-09-24 by jthermiz

alejoe91, 2020-09-24:
Good to hear that it's solved :)