SpikeInterface/spikeextractors #454: CacheRecordingExtractor: write_to_binary_dat_format() got an unexpected keyword argument 'verbose'

llobetv, 2020-09-30:
Hi,

When I do this I got an error:

```
rec, sort = se.example_datasets.toy_example(dump_folder=folder, dumpable=True, duration=10, num_channels=8)
rec = se.CacheRecordingExtractor(rec)
```

It appear that CacheRecordingExtractor call l.29

```
recording.write_to_binary_dat_format(save_path=self._tmp_file, dtype=self._dtype, chunk_size=chunk_size,
                                             chunk_mb=chunk_mb, verbose=verbose)
```
when verbose is not a possible argument : 
` write_to_binary_dat_format(self, save_path, time_axis=0, dtype=None, chunk_size=None, chunk_mb=500)`

Cheer,
Victor

alejoe91, 2020-10-01:
Hi @llobetv 

It is a possibel argument! https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/recordingextractor.py#L681

llobetv, 2020-10-01:
Not here: https://github.com/SpikeInterface/spikeextractors/blob/e589505bf0908352d1a16c506146acffe94117b0/spikeextractors/extractors/mdaextractors/mdaextractors.py#L58

This code:

```
import pytest
import spikeextractors as se
from spiketoolkit.postprocessing import export_to_phy
import os
import shutil

@pytest.mark.implemented
def test_export_to_phy():
    folder = 'test'
    if os.path.isdir(folder):
        shutil.rmtree(folder)
    rec, sort = se.example_datasets.toy_example(dump_folder=folder, dumpable=True, duration=10, num_channels=8)

    rec_Cache = se.CacheRecordingExtractor(rec)
    
    export_to_phy(rec_Cache, sort, output_folder='phy')#, copy_binary=True)  

if __name__ == '__main__':
    test_export_to_phy()

```

Give me:

> 
>   File "/users/nsr/llobet/Documents/spikeinterface/spiketoolkit-master_copy_binary/spiketoolkit/tests/test_postprocessing.py", line 32, in <module>
>     test_export_to_phy()
> 
>   File "/users/nsr/llobet/Documents/spikeinterface/spiketoolkit-master_copy_binary/spiketoolkit/tests/test_postprocessing.py", line 22, in test_export_to_phy
>     rec_Cache = se.CacheRecordingExtractor(rec)
> 
>   File "/users/nsr/llobet/miniconda2/envs/spyder3/lib/python3.7/site-packages/spikeextractors-0.9.2-py3.7.egg/spikeextractors/cacheextractors.py", line 30, in __init__
>     chunk_mb=chunk_mb, verbose=verbose)
>    
>    TypeError: write_to_binary_dat_format() got an unexpected keyword argument 'verbose'

alejoe91, 2020-10-01:
I see! I'll that's specific to mda. I'll fix it 🙂 thanks for catching it

alejoe91, 2020-10-01:
fixed https://github.com/SpikeInterface/spikeextractors/pull/455

closed 2020-10-01 by alejoe91