SpikeInterface/spikeextractors #464: Questions about BinDatRecordingExtractor data format and its .prb file acoordination

alexmaehon, 2020-10-13:
I turn Plexon .pl2 file into (channel, timeseries) type 'int16' binary .dat file, while success uesing BinDatRecordingExtractor(recording, sample_frequence=40000, numchan=256, dtpye='int16'), everything works fine untill run_sorters, most sorter fail.

Meanwhile, running tutorial data and example in spiketutorial this September file works fine, everything goes well. 

The only difference is BinDatRecordingExtractor and probe file, how should I format my data and probe file?

PS:
I use https://github.com/madeleinea/KilosortPLX to convert .pl2 file to .dat file. And binary file alone works fine in kilosort2 on Matlab, also works well when modified default .prb file channel number in Ironclus on Matlab.

A Newcomer to spikeinterface, wondering...

alexmaehon, 2020-10-13:
[ipynb_file.zip](https://github.com/SpikeInterface/spikeextractors/files/5369532/ipynb_file.zip)

alexmaehon, 2020-10-13:
This is  .ipynb file I use to run two scripts with only data and extractor difference.
(also .prb file)

alejoe91, 2020-10-13:
Hi @alexmaehon 

Thanks for asking!

Note that you can load the plexon data directly into SpikeInterface with the `PlexonRecordingExtractor` :)

From the output of your notebook, it seems that almost all sorters fail on the Pleson dataset. In particular, looking at KS2 output, it doesn't seem to find enough spikes. Do you see a lot of spiking activity in your original data? Maybe you could try another dataset with higher spiking activity?

Cheers 
Alessio

alexmaehon, 2020-10-13:
Thanks for your quick response.
I tried PlexonRecordingExtractor once, it did not work. 

I find that neoExtractor document says only works for old Plexon format .plx, and I try to turn .pl2 file into .plx using PlexUtil, and use PlexonRecordingExtractor to extract file:
"
Exception ignored in: <function BaseExtractor.__del__ at 0x0000024F8670ED38>
Traceback (most recent call last):
  File "...\spikeextractors\baseextractor.py", line 35, in __del__
AttributeError: 'PlexonRecordingExtractor' object has no attribute '_memmap_files'
Traceback (most recent call last):

  File "<ipython-input-7-ff5bd3d76307>", line 1, in <module>
    recording = se.PlexonRecordingExtractor('.plx')

TypeError: __init__() takes 1 positional argument but 2 were given
"

Plus,  .pl2 file is much faster than PLX file
                       PLX        PL2
SPK channel:  275.0s     0.18s
WB channel:   275.7s     2.50s
FP channel:     275.5s     0.08s

alexmaehon, 2020-10-13:
I just wondering what extra information needed to accompany with .dat file (channel, time-series)

alexmaehon, 2020-10-13:
> Hi @alexmaehon
>
> Thanks for asking!
>
> Note that you can load the plexon data directly into SpikeInterface with the `PlexonRecordingExtractor` :)
>
> From the output of your notebook, it seems that almost [...]

Thanks for your quick response.
I tried PlexonRecordingExtractor once, it did not work.

I find that neoExtractor document says only works for old Plexon format .plx, and I try to turn .pl2 file into .plx using PlexUtil, and use PlexonRecordingExtractor to extract file:

""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""
Exception ignored in: <function BaseExtractor.del at 0x0000024F8670ED38>
Traceback (most recent call last):
File "...\spikeextractors\baseextractor.py", line 35, in del
AttributeError: 'PlexonRecordingExtractor' object has no attribute '_memmap_files'
Traceback (most recent call last):

File "", line 1, in
recording = se.PlexonRecordingExtractor('.plx')

TypeError: init() takes 1 positional argument but 2 were given
""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""

Plus, .pl2 file is much faster than PLX file:
In tests, a file containing ten minutes of spike (SPK), wideband (WB), and field potential (FP) data from 256 channels was recorded in both PLX (13.8GB file) and PL2 (11.4GB file) formats.

PLX file:
SPK channel: 275.0s
WB channel: 275.7s
FP channel: 275.5s

PL2 file:
SPK channel: 0.18s
WB channel: 2.50s
FP channel: 0.08s

I just wondering what extra information needed to accompany with .dat file (channel, time-series)

Indeed, this data might only contain 2 to 3 channel has spikes out of 58.
Currently, there is some issue with this animal that prevents further data acquisition. I will try to find some example .plx file, and try it again.

alejoe91, 2020-10-13:
The information needed by the `BinDatRecordingExtractor` are:
- `numchan`
- `sampling_frequency`
- `dtype`

You can check how the signals look with the `plot_timeseries` and `plot_spectrum` (I see that you have it already). The data look correctly loaded to me so I really think that the issue is a lack of spikes

samuelgarcia, 2020-10-13:
The signature of PlexonRecordingExtractor is
```PlexonRecordingExtractor(filename='myfilename.plx')```
could you try it ?

alejoe91, 2020-10-13:
@samuelgarcia it's a `.pl2` file, not supported bu neo. Or is it supported now?

samuelgarcia, 2020-10-13:
I think it is not, I need to check.