SpikeInterface/spikeextractors #475: concatenate multiple recording files

gmaggi, 2020-10-21:
Hi,

Is there any facility in spikeinterface to concatenate multiple recording files? thus, the concatenated output file is the one that would be actually passed to a sorter.
I see this doc: https://djoshea.github.io/neuropixel-utils/imec_dataset/#concatenating-multiple-files-together and I wonder if something like this is already integrated.

Thanks,
Giuliano.

alejoe91, 2020-10-21:
Hi @gmaggi 

Of course! :) 

If you have a list of recordings `rec_list`, you can just run:

`multirec = se.MultiRecordingTimeExtractor(rec_list)`

The output `multirec` is also a `RecordingExtractor`, so you can directly sort it.

Cheers
Alessio

gmaggi, 2020-10-21:
Thank you @alejoe91 ,

Another question, how could one handle x_pitch and y_pitch (for glx) using `se.MultiRecordingTimeExtractor(rec_list)`  ?Currently for a single file we do it as `se.SpikeGLXRecordingExtractor(input_file, x_pitch, y_pitch)` .

-Giuliano.

alejoe91, 2020-10-21:
Hi @gmaggi 

Also x_pitch and y_pitch do not need to be passed to the constructor, as they are by default correct for the Neuropixels probe and locations are read correctly.

Of course concatenation assumes that all locations are the same (you used the same configuration for all recordings). 

So what you would do is the following:

```
# load list of spikeGLX files
spike_glx_files = ['file1,ap.bin', 'file2.ap.bin' ...]
rec_list = []
for f in spike_glx_files:
    rec = se.SpikeGLXRecordingExtractor(f)
    rec_list.append(rec)

# create concatenated recording
multirec = se.MultiRecordingTimeExtractor(rec_list)
```

Let me know if you run into issues!

gmaggi, 2020-10-21:
Ok, I let you know @alejoe91 
Another related question, when we manage a single file, it is done as:

```
recording = se.SpikeGLXRecordingExtractor(single_input_file)
subrecording = recording.load_probe_file(prb_file)

```
Is the load_probe_file needed?, or is it automatically done from the ap.meta file?

Thanks,
Giuliano.

alejoe91, 2020-10-21:
For neuropixels it's not needed! You can check the loactions with:

```
import spikewidgets as sw

w = sw.plot_electrode_geometry(recording)
```

gmaggi, 2020-10-27:
Hi @alejoe91 , all worked fine with se.MultiRecordingTimeExtractor as you proposed.
Thanks,
Giuliano.

alejoe91, 2020-10-27:
Awesome! Closing the issue then

closed 2020-10-27 by alejoe91