SpikeInterface/spikeextractors #476: AttributeError: module 'spikeforest2_utils' has no attribute '__version__'

timsainb, 2020-10-23:
This is a cross-post of https://github.com/flatironinstitute/spikeforest2/issues/28 because I'm not sure where it is more appropriate, here or spikeforest2.

I received this error, running st.validation.compute_quality_metrics on spikeforest2 data. 

Here is a colab notebook reproducing the error: https://colab.research.google.com/drive/14czFArWXjU6TjnLRc927231pY2qE5NGC?usp=sharing 

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-13-75359faf695c> in <module>()
      3     recording=recording,
      4     metric_names=["firing_rate", "isi_violation", "snr", "nn_hit_rate", "nn_miss_rate"],
----> 5     as_dataframe=True,
      6 )

5 frames
/usr/local/lib/python3.6/dist-packages/spikeextractors/baseextractor.py in make_serialized_dict(self)
     57         if self.is_dumpable:
     58             dump_dict = {'class': class_name, 'module': module, 'kwargs': self._kwargs,
---> 59                          'key_properties': self._key_properties, 'version': imported_module.__version__,
     60                          'dumpable': True}
     61         else:

AttributeError: module 'spikeforest2_utils' has no attribute '__version__'
```

alejoe91, 2020-10-23:
Hi! This is a duplicate of this issue:

https://github.com/SpikeInterface/spikeinterface/issues/80

The reason is that spikeforest_utils does not have a `__version__`, which is required by SpikeInterface. This is because SI objects, when processed in parallel, are dumped to a dictionary which stores the version of the package that created it.

@magland can you fix this?

timsainb, 2020-10-23:
Thanks Alex, for the time being, I came up with a workaround by converting both everything the numpyextractor format:

https://colab.research.google.com/drive/19_G_nrGOjqrM3yZSBP77_fYU2wa9H4BX?usp=sharing

alejoe91, 2020-10-23:
Hi @timsainb 

Why do you access the `get_traces()` as `recording._recording.get_traces()`? All the methods should be accessible directly as `recording.get_traces()`. Usually the fields starting with `_` are supposed to be private :)

Note that your approach will try to load all recordings in RAM! Instead, I recommend caching the recording and sorting extractors. This will create a binary copy of the traces and all metadata to disk, without loading them into memory:

```
recording_cache = se.CacheRecordingExtractor(recording, save_path='where-you-want-to-save-it.dat')
sorting_cache = se.CacheSortingExtractor(sorting_true, save_path=''where-you-want-to-save-it.npz')
```

Then you can proceed using these two objects (`recording_cache` and `sorting_cache`). Sounds good?

timsainb, 2020-10-23:
That works. Thanks @alejoe91 !

closed 2021-03-18 by alejoe91