SpikeInterface/spikeextractors #480: SpikeExtractor no longer works (groupdict error)

ohnopleasehelpme, 2020-10-29:
I am simply trying to extract a .ncs filetype for tetrode data. I am getting this same error on both my mac and PC.

import spikeinterface.extractors as se
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw
import matplotlib.pylab as plt
import numpy as np
import neo

reader = neo.NeuralynxIO(dirname='~/PycharmProjects/SpikeGUI/data')
print(reader)
print(reader.segment_count(0))

rec_list = [se.NeuralynxRecordingExtractor(dirname='~/PycharmProjects/SpikeGUI/data/', seg_index=i)
            for i in range(reader.segment_count(0))]
rec = se.MultiRecordingTimeExtractor(rec_list)

Error Message:

runfile('/Users/~/PycharmProjects/SpikeGUI/Neuralynx_trials.py', wdir='/Users/~/PycharmProjects/SpikeGUI')
/Users/~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/elephant/pandas_bridge.py:22: DeprecationWarning: pandas_bridge module will be removed in Elephant v0.8.x
  DeprecationWarning)
00:28:36 [I] klustakwik KlustaKwik2 version 0.2.6
Backend MacOSX is interactive backend. Turning interactive mode on.
Traceback (most recent call last):
  File "/Users/~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-9b1077401436>", line 1, in <module>
    runfile('/Users/~/PycharmProjects/SpikeGUI/Neuralynx_trials.py', wdir='/Users/~/PycharmProjects/SpikeGUI')
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/Users/~/PycharmProjects/SpikeGUI/Neuralynx_trials.py", line 12, in <module>
    reader = neo.NeuralynxIO(dirname=neuralynx_folder)
  File "/Users/~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/io/neuralynxio.py", line 36, in __init__
    BaseFromRaw.__init__(self, dirname)
  File "/Users/~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/io/basefromrawio.py", line 81, in __init__
    self.parse_header()
  File "/Users/~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/rawio/baserawio.py", line 151, in parse_header
    self._parse_header()
  File "/Users/~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/rawio/neuralynxrawio.py", line 98, in _parse_header
    info = read_txt_header(filename)
  File "/Users/~/opt/miniconda3/envs/SpikeGUI/lib/python3.7/site-packages/neo/rawio/neuralynxrawio.py", line 682, in read_txt_header
    dt1 = re.search(datetime1_regex, txt_header).groupdict()
AttributeError: 'NoneType' object has no attribute 'groupdict'

colehurwitz, 2020-11-12:
Hi @ohnopleasehelpme ,

This looks like an issue with the rawio capabilities of the NEO package. Any ideas about this @samuelgarcia ?

samuelgarcia, 2020-11-12:
@ohnopleasehelpme : yes, it is neo.rawio.neuralynxrawio related.
Could you open an ussir in https://github.com/NeuralEnsemble/python-neo/issues
And maybe you could ask specifically to @PeterNSteinmetz and/or @JuliaSprenger
they known quite well this format.

PeterNSteinmetz, 2020-11-12:
@ohnopleasehelpme Concur with samuelgarcia that this is likely an issue with the neuralynxrawio. There are a number of specific configurations of the file headers and record structures which are not presently well supported.

Suggest opening the issue at the location suggested and including at least one of the ncs or ntt files which is causing the troubles with that directory or at least a truncated portion of one file. You could check if that creates the same error when isolated in a directory. You can truncate a file to a length of 16384 + 1044 x N bytes, where N is a whole number of records, and it should still be a valid file. Smaller test files are a bit easier to deal with.

ohnopleasehelpme, 2020-11-17:
I see - @PeterNSteinmetz I will try your suggestion and possibly try pip reinstalling neo - this was working not 2 months ago so I am not sure what has changed.