SpikeInterface/spikeextractors #494: Suggested enhancement for calculating sampling rates for NWB files

lfrank, 2020-11-13:
Currently sampling rates are calculated as 1/np.median(np.diff(timestamps)). There are circumstances where this will be incorrect, as when a set of timestamps are invalid (e.g. nan). The following code provides a somewhat more robust sampling rate calculation. I typically use a multiplier of 1.5.

```
def estimate_sampling_rate(timestamps, multiplier):
    '''Estimate the sampling rate given a list of timestamps. Assumes that the most common temporal differences
       between timestamps approximate the sampling rate. Note that this can fail for very high sampling rates and
       irregular timestamps
    :param timestamps: 1D numpy array (list of timestamps)
    :return: estimated_rate: float
    '''

    #approach:
    # 1. use a box car smoother and a histogram to get the modal value
    # 2. identify adjacent samples as those that have a time difference < the multiplier * the modal value
    # 3. average the time differences between adjacent samples
    sample_diff = np.diff(timestamps[~np.isnan(timestamps)])
    if len(sample_diff) < 10:
        print(f'Error in estimate_sampling_rate: only {len(sample_diff)} timestamps are valid. Check the data.')
        return -1
    nsmooth = 10
    smoother = np.ones(nsmooth) / nsmooth
    smooth_diff = np.convolve(sample_diff, smoother, mode='same')

    # we histogram with 100 bins out to 3 * mean, which should be fine for any reasonable number of samples
    hist, bins = np.histogram(smooth_diff, bins=100, range=[0, 3 * np.mean(smooth_diff)])
    mode = bins[np.where(hist == np.max(hist))]

    adjacent = sample_diff < mode[0] * multiplier
    return np.round(1.0 / np.mean(sample_diff[adjacent]))
```

alejoe91, 2020-11-13:
Hi @lfrank 

Thanks for the suggestion :) @bendichter @CodyCBakerPhD could one of you guys add this to the `NwbRecordingExtractor`?

bendichter, 2020-11-13:
@lfrank Interesting approach. Seems to be similarly robust to outliers, and might be faster too, since the `median` operation can be slow. I don't recall ever seeing NaN timestamps, and I'm curious where they would occur.

samuelgarcia, 2020-11-13:
@lfrank :
I anderstand the smoothing but I not sure to follow you on why the modal (with max on hist) is better that a median with the same smooth.

lfrank, 2020-11-13:
So … the main reason is that in some cases, there can be a delay between having a sample come in to the computer and having it timestamped. This gets complicated, but in cases where samples can be lost (which does happen), one may need a separate clock. 

We’re using unix time for all timestamps, as that tells us exactly when each sample was collected, but since our data acquisition system doesn’t directly encode unix time, there’s a lookup process involved, and at very high data speeds that can choppy, such that, for example, there can be an alternation between larger and smaller times, where (larger + smaller) /  2 is very close to the real sample interval. We of course correct for that, but just in case that correction isn’t perfect, I think that taking modal value of the smoothed data will (I believe) come closest to finding the value, while the median only works correctly if there are no missing samples.

Loren

> 
> 
> @lfrank <https://github.com/lfrank> :
> I anderstand the smoothing but I not sure to follow you on why the modal (with max on hist) is better that a median with the same smooth.