SpikeInterface/spikeextractors #515: Index error on KS

Levi55555, 2020-12-05:
I got an Index error when I tried to run my code on KS data.

`import spikeextractors as se
import spiketoolkit as st

sorting = se.KiloSortSortingExtractor('ARat_2017_09_11_128ch_probe_PD_170911_132447')
recording = st.preprocessing.bandpass_filter(recording)
waveforms = st.postprocessing.get_unit_waveforms(recording, sorting, max_spikes_per_unit=1000, chunk_mb=2000, n_jobs=8)
available_metrics = st.validation.get_quality_metrics_list()
metrics_df = st.validation.comute_metrics(sorting, recording, metric_names=available_metrics, `as_dataframe=True)``

Error message:
Traceback (most recent call last):
  File "Spikeinterface.py", line 9, in <module>
    sorting = se.KiloSortSortingExtractor('ARat_2017_09_11_128ch_probe_PD_170911_132447')
  File "/home/balazsi/.local/lib/python3.6/site-packages/spikeextractors/extractors/kilosortextractors/kilosortextractors.py", line 26, in __init__
    PhySortingExtractor.__init__(self, folder_path, exclude_cluster_groups, load_waveforms, verbose)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spikeextractors/extractors/phyextractors/phyextractors.py", line 95, in __init__
    property_name = tokens[1]
IndexError: list index out of range

alejoe91, 2020-12-05:
Hi,

The `quality_metrics.csv` is not a KS-generated file. The extractor attempts to read all tsv and csv files to load properties. I think that this is messing things up. Can you try again after moving that file?

Levi55555, 2020-12-05:
True, I have removed that file, and I got another error:
Traceback (most recent call last):
  File "Spikeinterface.py", line 8, in <module>
    recording = KiloSortRecordingExtractor('ARat_2017_09_11_128ch_probe_PD_170911_132447')
  File "/home/balazsi/.local/lib/python3.6/site-packages/spikeextractors/extractors/kilosortextractors/kilosortextractors.py", line 14, in __init__
    PhyRecordingExtractor.__init__(self, folder_path)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spikeextractors/extractors/phyextractors/phyextractors.py", line 33, in __init__
    BinDatRecordingExtractor.__init__(self, datfile[0], sampling_frequency=float(self.params['sample_rate']),
IndexError: list index out of range

alejoe91, 2020-12-05:
It seems it doesn't find the dat file. Is it still.in the folder? Can you check that the path in the params.py is correct?

Levi55555, 2020-12-09:
Hi!
The missing .dat file has occured the issue it has been solved.
I tried several parameters to run the script but at every try I got this:

"
Traceback (most recent call last):
  File "Spikeinterface.py", line 13, in <module>
    waveforms = st.postprocessing.get_unit_waveforms(recording, sorting, max_spikes_per_unit=1000, chunk_mb=4000, n_jobs=8)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spiketoolkit/postprocessing/postprocessing_tools.py", line 281, in get_unit_waveforms
    for ii in chunk_iter)
ValueError: too many values to unpack (expected 1)
"
Our .dat is quite big (almost 19Gb) I have 40Gb free space on my SSD therefore I moved the KS folder to an external HDD and I tried to run the python script from there. Could it be a problem?

alejoe91, 2020-12-09:
Hi Levi,

We are currently releasing a new version of all the packages with a few bug fixes (and I think that would solve your issue too)!

Can you upgrade all the packages and try again?
`pip install --upgrade spikeectractors spiketoolkit spikesorters spikecomparison spikewidgets`

No problem if the file is big! Everything is done in chunks and by default only 500Mb of RAM will be used. If you have more RAM you can use the `chunk_mb=2000` (e.g. for using 2Gb of RAM) as an additional argument to the `get_unit_waveforms`

Levi55555, 2020-12-09:
The new version has fixed the bugs but...

"
Traceback (most recent call last):
  File "Spikeinterface.py", line 18, in <module>
    metrics_df = st.validation.compute_quality_metrics(sorting, recording, metric_names=available_metrics, as_dataframe=True)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spiketoolkit/validation/quality_metrics.py", line 1220, in compute_quality_metrics
    md.compute_amplitudes(**kwargs)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spiketoolkit/validation/quality_metric_classes/metric_data.py", line 186, in compute_amplitudes
    self._recording, self._sorting, **kwargs)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spiketoolkit/validation/quality_metric_classes/utils/validation_tools.py", line 157, in get_amplitude_metric_data
    spike_clusters_amp, amplitudes = _get_amp_metric_data(recording, sorting, **kwargs)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spiketoolkit/postprocessing/postprocessing_tools.py", line 1496, in _get_amp_metric_data

"

:(

alejoe91, 2020-12-09:
Can you paste the full error?

Levi55555, 2020-12-09:
Traceback (most recent call last):
  File "Spikeinterface.py", line 18, in <module>
    metrics_df = st.validation.compute_quality_metrics(sorting, recording, metric_names=available_metrics, as_dataframe=True)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spiketoolkit/validation/quality_metrics.py", line 1220, in compute_quality_metrics
    md.compute_amplitudes(**kwargs)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spiketoolkit/validation/quality_metric_classes/metric_data.py", line 186, in compute_amplitudes
    self._recording, self._sorting, **kwargs)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spiketoolkit/validation/quality_metric_classes/utils/validation_tools.py", line 157, in get_amplitude_metric_data
    spike_clusters_amp, amplitudes = _get_amp_metric_data(recording, sorting, **kwargs)
  File "/home/balazsi/.local/lib/python3.6/site-packages/spiketoolkit/postprocessing/postprocessing_tools.py", line 1496, in _get_amp_metric_data
    amplitudes[i_start_amp:i_start_amp + len(st_amp)] = amp[:, np.newaxis]
ValueError: could not broadcast input array from shape (4853,1,1) into shape (4853,1)

alejoe91, 2020-12-09:
Can you try to add `recompute_info=True` as an argument to the compute metrics function? I think the error could be due to the fact the amplitudes are loaded from kilosort..but we have to investigate that

alejoe91, 2020-12-10:
@Levi55555 I found the bug. Basically the amplitudes are loaded with an extra dimension (e.g. (1500, 1)), and that messes up the compute metrics function.

Can you try to install spikeextarctors from this PR: https://github.com/SpikeInterface/spikeextractors/pull/521 ?

It should fix that :)

closed 2020-12-11 by alejoe91