SpikeInterface/spikeextractors #517: How to spike sort separate recordings together?

NilsNyberg, 2020-12-06:
Hi everyone,

I love what I have seen of SpikeInterface so far, and am in the progress of building my own pipeline. However, I am not sure how to spike sort separate recordings together.

The pipeline I am interested in creating is as follows: in one recording session, my rats perform 4 different blocks of a navigation task, and between these blocks are 4 short sleeping blocks. In total, this gives 8 separate recordings that I want to spike sort together. I am recording from a total of 32 tetrodes, 16 in each hemisphere. I therefore want to spike sort the data from all of the 8 session blocks together, but separately for each hemisphere.  Can this be done? Combined, the files are quite large, up towards 300gb. Lastly, at the end of spike sorting and manual curation I would like to output each session data into either one big NWB file (containing all the session blocks), or separate NWB files for each blocks. Can spikeInterface handle this as well?

Thanks for the help!

alejoe91, 2020-12-06:
Hi @NilsNyberg 

Sure that is possible. I would use a code like this:

```
import spikeextractors as se
import spikesorters as ss

recordings = []

for oe_folder in open_ephys_folders:
    rec = se.OpenEphysRecordingExtractor(oe_folder)
    recordings.append(rec)

multirec = se.MultiRecordingTimeExtractor(recordings)

# Since you have tetrodes, you can add grouping info with a probe file or the set_channel_groups() function

multirec.set_channel_groups([0,0,0,0,1,1,1,1 ...])

# you can then sort the whole thing (by group):
multisort = ss.run_mountainsort4(multirec, grouping_property='group')

# then split the sorted data
sortings = []
for epoch in multirec.get_epoch_names():
    info = multirec.get_epoch_info(epoch)
    sort = se.SubSortingExtractor(multisorting, start_frame=info['start_frame'], end_frame=info['end_frame'])
    sortings.append(sort)
```

I plan to add something like this to the docs in the next days.

Cheers
Alessio

NilsNyberg, 2020-12-06:
Awesome! Thanks! I might come back with more questions if I run into any.

Best,
Nils

NilsNyberg, 2020-12-07:
I haven't had a chance to try it yet, but is it important that the timestamps across the files are consecutive? Since I re-name each file in Open Ephys before starting the recording for each of my 8 navigation/sleeping blocks , I think all of the 8 files will have timestamps starting from 0 (i.e., and not follow from the where the previous recording ended).

alejoe91, 2020-12-08:
The 8 files will be concatenated as one (so the start time is automatically computed for each segment). This might cause an artifact at the interface, which you can zero out with the `remove_artifacts` preprocessor

alejoe91, 2020-12-10:
@NilsNyberg I added an example in the docs to handle concatenation of recordings: https://spikeinterface.readthedocs.io/en/latest/modules/sorters/plot_4_sorting_concatenated_recordings.html#sphx-glr-modules-sorters-plot-4-sorting-concatenated-recordings-py

closed 2020-12-10 by alejoe91