SpikeInterface/spikeextractors #533: Neuralynx SpikeExtractors not working?

ohnopleasehelpme, 2020-12-30:
I am just running a simple code to extract a Neuralynx signal. This was working over the summer and then randomly stopped working so I am not sure if the issue is on my end. 

`
import spikeinterface.extractors as se
import spikeinterface.toolkit as st
import spikeinterface.sorters as ss
import spikeinterface.comparison as sc
import spikeinterface.widgets as sw
import matplotlib.pylab as plt
import numpy as np
import neo

neuralynx_folder = 'data/'
reader = neo.NeuralynxIO(dirname=neuralynx_folder)
print(reader.header['signal_channels']['units'])
print(reader.header['signal_channels']['units'].shape)

extractor = se.NeuralynxRecordingExtractor(dirname=neuralynx_folder, block_index=0, seg_index=0)
`
I am getting the following error:

`~/PycharmProjects/Spike2021/main.py
11:23:29 [I] klustakwik KlustaKwik2 version 0.2.6
['uV' 'uV' 'uV' 'uV']
(4,)
Traceback (most recent call last):
  File "~/PycharmProjects/Spike2021/main.py", line 15, in <module>
    extractor = se.NeuralynxRecordingExtractor(dirname=neuralynx_folder, block_index=0, seg_index=0)
  File "~.conda/envs/Spike2021/lib/python3.7/site-packages/spikeextractors/extractors/neoextractors/neobaseextractor.py", line 57, in __init__
    num_chan_group = len(self.neo_reader.get_group_channel_indexes())
AttributeError: 'NeuralynxRawIO' object has no attribute 'get_group_channel_indexes'

Process finished with exit code 1
`

Does this mean Neo is not running properly? I have both python 3.7 and 3.9.

alejoe91, 2020-12-31:
Hi, this should be fixed in master. Can you install from sources?

ohnopleasehelpme, 2020-12-31:
Hi Alessio - Master in Neo or SpikeInterface? Where can I download the updated version?

ohnopleasehelpme, 2021-01-01:
Hey Alessio - where can I install from sources?

alejoe91, 2021-01-01:
Hi! You can see the instructions at the bottom of this page in the documentation: https://spikeinterface.readthedocs.io/en/latest/installation.html

ohnopleasehelpme, 2021-01-02:
Hi Alessio - I get the same exact error...

migparra73, 2021-01-04:
I commented out 
`num_chan_group = len(self.neo_reader.get_group_channel_indexes())`
`assert num_chan_group == 1, 'This file have several channel groups spikeextractors support only one groups`
in neobaseextractor.py and it seems to work. Maybe a condition needs to be added to the NeoBaseExtractor to exclude neuralynx files from this check?

alejoe91, 2021-01-04:
Thanks @migparra73 

We already updated the `NEObaseextarctor` to deal with both the olde version (`<0.9.0`) and the new versions (`>=0.9.0`) (https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/neoextractors/neobaseextractor.py#L57-L62https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/neoextractors/neobaseextractor.py#L57-L62)

@ohnopleasehelpme what is the output of `pip list`? I'm pretty sure the latest master version was not installed correctly in your case...