SpikeInterface/spikeextractors #540: Multiple signals per channel on BlackrockRecordingExtractor

luiztauffer, 2021-01-13:
`BlackrockRecordingExtractor`, which uses `NeoBaseRecordingExtractor`, suffers from the problem (noted elsewhere in the code) of ambiguous channels when there are, multiple `ns*` files, containing signals sampled in different rates for the same channels.

I think this could be dealt with in two ways:
1. pass an extra argument to Neo telling it to map only one specific file. Is that a solution simple (and desirable) to be implemented in Neo? 
2. We can map the files indirectly with `BlackrockRecordingExtractor`, given that groups (files) differ in their sampling rate. This is more circumventing the problem and sounds like prone to future problems

@samuelgarcia @alejoe91 ideas?

samuelgarcia, 2021-01-13:
Hi @luiztauffer.
I known the problem.
neo handle several stream formats. here blackrock with ns6 + ns2 is a multi stream format.
In the wrapper, we should add a keywords to select which stream or which channel ids.

A major problem for is that the end user don't known in advance which stream to use and the semantic around it. (here ns6, ns2, ns5)

Maybe we could:
 * add a list_stream() function helper in neo wrapper
 * auto select the "best" stream for spike sorting which should be the higher sampler rate. like you propose.

For the very specific case of BlacrockIO neo have the `nsx_to_load = [6]` option that can solve your problem straight now.

samuelgarcia, 2021-01-13:
Note that I plan an improvement of the rawio APi of neo.
See https://github.com/NeuralEnsemble/python-neo/issues/895
When done I will improve wrapper on top of neo.
Not started yet

luiztauffer, 2021-01-13:
>> A major problem for is that the end user don't known in advance which stream to use and the semantic around it. (here ns6, ns2, ns5)

The user already chooses it actually, when passing `filename=filename.ns3`. 
Besides, it seems that Blackrock has the file extensions associated with fixed sampling rates, so it would be simple to return a print message to the user saying that “file X, with X Hz, is being used, if you want another file, use arg=value” and also “If you ‘re planning to run spike sorting, please use .ns6”

>>  For the very specific case of BlacrockIO neo have the nsx_to_load = [6] option that can solve your problem straight now.

Yes I think that would solve the problem. `BlackrockRecordingExtractor` right now uses rawIO, will we be lossing something by changing it?

samuelgarcia, 2021-01-13:
> The user already chooses it actually, when passing filename=filename.ns3.

the neo.BlacrockIo bypass this choice by loading all "connected" in the folder.

`BlackrockRecordingExtractor` forward all kwargs to neo rawio. So
`BlackrockRecordingExtractor(..., nsx_to_load=3)` should work.
Never tested but should work.

luiztauffer, 2021-01-13:
>> BlackrockRecordingExtractor(..., nsx_to_load=3) should work.

this seems to be working fine, thanks!
I'm closing this issue

PR #541 (2021-01-21): Add return_scaled to RecordingExtractor get_traces()
fix #540
closed 2021-01-13 by luiztauffer