SpikeInterface/spikeextractors #542: Phy opening issue and write sorting with channel/group info

Clementp69, 2021-01-14:
Hi, 

I am using SpikeInterface to sort spikes from my 16 channel recordings. 

I've configured my probe file in order to have one channel per group and I sorted spikes with sorters by adding 
 > grouping_property='group'. 

This doing 16 sorting one per channel. 
My sorting is okay but when I want to export it on phy with these command :
> st.postprocessing.export_to_phy(recording, sorting_IRON, output_folder='sorter140121', grouping_property='group')
> !phy template-gui sorter140121/params.py

I have this error : 
11:58:55.853 [E] __init__:62          An error has occurred (AssertionError): 
Traceback (most recent call last):
  File "c:\program files\anaconda3\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\program files\anaconda3\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Anaconda3\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\program files\anaconda3\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\program files\anaconda3\lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "c:\program files\anaconda3\lib\site-packages\phy\apps\template\gui.py", line 198, in template_gui
    controller = TemplateController(model=load_model(params_path), dir_path=dir_path, **kwargs)
  File "c:\program files\anaconda3\lib\site-packages\phylib\io\model.py", line 1178, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "c:\program files\anaconda3\lib\site-packages\phylib\io\model.py", line 297, in __init__
    self._load_data()
  File "c:\program files\anaconda3\lib\site-packages\phylib\io\model.py", line 356, in _load_data
    self.sparse_templates = self._load_templates()
  File "c:\program files\anaconda3\lib\site-packages\phylib\io\model.py", line 652, in _load_templates
    assert cols.shape == (n_templates, n_channels_loc)
AssertionError

But it is working when I remove this parameter from the export :
>grouping_property='group'

Does someone have a solution to this? 

Also, when I save the results of my sorting I don't have the info of the channel or group in the saved file, I am using this command : 
> se.NpzSortingExtractor.write_sorting(sorting=sorting_IRON, save_path='write_sorting140121.npz')

Which parameters must be added to have this info in the saved file? 

Thanks in advance, 
Clément

alejoe91, 2021-01-14:
Hi Clement,

Do you want to spike sort each channel separately? Are the electrodes far apart? In that case, you are correctly using the `grouping_property='group` argument. I have never tested the `export_to_phy` with groups of only one channel, I'll take a look at it in the next days.

In order to save an extractor along with properties and features, you can use the `dump_to_pickle()` function:

```
sorting.dump_to_pickle("my-sorted-data.pkl", include_properties=True)

# to load back:
sorting_loaded = se.load_extractor_from_pickle("my-sorted-data.pkl")
```

Clementp69, 2021-01-14:
Hi, 

Yes, I would like to sort each channel separately because each of the 16 channels is spaced by 200µm. Thanks for taking a look, 

I tried to play with this function, but I don't succeed to extract values info in R or Matlab. I don't see when I import it on R, the channel/group information. The idea was to add for instance to add the units_map in the NpzSortingExtractor. 

I'll try to save it "unit_map" separately, thanks.

alejoe91, 2021-01-15:
Hi Clement, you can try to save the output to NWB (with the NWBSortingExtractor.write_sorting()). This saves all the unit properties (including group) and there is a matlab API called matNWB to read the file in matlab. I think it could be a viable solution on your case!

alejoe91, 2021-01-15:
So, I also got the same error when using single-channel groups..I think it might be due to some squeezing of arrays done in `phy` or `phylib`. Maybe @rossant can help? 

I can also try to take a look in the next weeks :)

Clementp69, 2021-01-17:
Hi Alessio, 

Yes thank you for your answers. I explain you what i've done. 

Before using NPZ to save my sorting, I tried to save it with "NWBSortingExtractor.write_sorting()". It worked but when i loaded in matlab, the file had all the folders but empty. So i tried another extractor. 

I tried to play a lot with yours scripts but since now my matlab file is still empty. I don't know if this writing proble is due to an issue in the extractors script or in matlab packages. 

Have you tried to import your nwb sorting file in matlab ? 

For the importation issue in phy, i'll try to contact @rossant.

Best, 
Clément