SpikeInterface/spikeextractors #543: BlackrockSortingExtractor error when reading .nev file

luiztauffer, 2021-01-15:
When trying to use `BlackrockSortingExtractor` to read spikes from a `.nev` file:
```python
from spikeextractors import BlackrockSortingExtractor

se = BlackrockSortingExtractor(filename='file.nev')
```

This yields an error:
```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-15-5c65512a0164> in <module>
      4 
      5 datafile = '/media/luiz/storage/taufferconsulting/client_ben/project_movshon/data/XX_LE_textures_20191128_002.nev'
----> 6 se = BlackrockSortingExtractor(filename=datafile)

/media/luiz/storage/Github/spikeextractors/spikeextractors/extractors/neoextractors/neobaseextractor.py in __init__(self, **kargs)
    150         # neo handle this but not spikeextractors
    151 
--> 152         self._handle_sampling_frequency()
    153 
    154     def _handle_sampling_frequency(self):

/media/luiz/storage/Github/spikeextractors/spikeextractors/extractors/neoextractors/neobaseextractor.py in _handle_sampling_frequency(self)
    160         #  all channels are in the same neo group so
    161         self._neo_sig_sampling_rate = self.neo_reader.header['signal_channels']['sampling_rate'][0]
--> 162         self._neo_sig_time_start = self.neo_reader.get_signal_t_start(self.block_index, self.seg_index,
    163                                                                       channel_indexes=None)
    164 

~/anaconda3/envs/env_movshon/lib/python3.8/site-packages/neo/rawio/baserawio.py in get_signal_t_start(self, block_index, seg_index, channel_indexes)
    460     def get_signal_t_start(self, block_index, seg_index, channel_indexes=None):
    461         if self._several_channel_groups:
--> 462             self._check_common_characteristics(channel_indexes)
    463         return self._get_signal_t_start(block_index, seg_index, channel_indexes)
    464 

~/anaconda3/envs/env_movshon/lib/python3.8/site-packages/neo/rawio/baserawio.py in _check_common_characteristics(self, channel_indexes)
    396         # ~ print('_check_common_characteristics', channel_indexes)
    397 
--> 398         assert channel_indexes is not None, \
    399             'You must specify channel_indexes'
    400         characteristics = self.header['signal_channels'][_common_sig_characteristics]

AssertionError: You must specify channel_indexes
```

samuelgarcia, 2021-01-15:
I don't remember exactly.
The main idea is that neo spiketrain don't have sampling rate. And SortingExtractor need it.
neo deal directly with spike times as s.
So I need to link theses spikes to "signal" stream to catch the sampling rate.
I need a stronger insight on it. I will have a look next week.

luiztauffer, 2021-01-15:
@samuelgarcia it looks like the method `self.neo_reader.get_signal_t_start()` needs to receive a list of int for the `channel_indexes` arg

samuelgarcia, 2021-01-15:
And again, as I previously say to Ben.
If you use neo base extractor for sorting then OK, we need to fix it.
But If you use neo based axtractor for format convertion, I think this is a bad idea. really.

You are fighting with an API reduction between neo and spikeinterface.
The neo API for data description is far richer than the spikeinterface API.

For instance here you should not read spike in the nev with the BlackrockSortingExtractor but rather with neo.
This would give you times (and not frame). Then you could make a custimised extractor based on numpy (quantities in fact) vectors.

samuelgarcia, 2021-01-15:
Yes. You find it. this is it.
But it will works only in the case unique stream. (one nsx)

luiztauffer, 2021-01-15:
> Yes. You find it. this is it.
> But it will works only in the case unique stream. (one nsx)

why wouldn't it work with multiple streams? Their initial t should be the same, just sampling frequency changes, right?

samuelgarcia, 2021-01-15:
yes it is about the sampling rate when multiple signal stream it is hard to "guess" wich stream spike belong to.
So times to frames is more complicated than it appear.
This could be fixed in at neo level by using annotation dict and adding some special keys usefull for spikeinterface extractor to help for this king of details.

luiztauffer, 2021-02-02:
passing `nsx_to_load` to the sorting extractor, just like the recording extractor, seems to help. At least when I do this, the `get_sampling_frequency()` returns the correct frequency of the chosen stream.
do you think explicitly passing `nsx_to_load` to the sorting extractor solves the issue of stream ambiguity, @samuelgarcia ?

samuelgarcia, 2021-02-02:
Yes. In the context of blackrock system nsx_to_load is equivalent to select a unique stream. (unless you give a list).
For the sorting in actual implementation it should also work. But the implementaion is quite bad now.
The main problem is that neo deal with spike times in second (or with timestamp that are not necessary at the sampling rate we want) but spikeinterface deal with spike sample relative to a recording (which is generally the case).
The neo SpikeTrain don't need this link to signals because it is time.

In the next generation of spikeinterface  https://github.com/SpikeInterface/spikeinterface/pull/88, I am doing something better for sorting based on neo.

luiztauffer, 2021-02-02:
Thanks for clarifying it!
The main reason to insist with the SpikeInterface BlackrockExtractors to deal with Blackrock, instead of just going straight to NEO, is because we already have good standards in place to interface NWB converters with SpikeInterface, but not yet with NEO.

The refactor looks interesting! Do you expect the SpikeInterface api to change much after that?

samuelgarcia, 2021-02-02:
The refactoring at the moment it just a proposal to the team.
I can make comments if you want of course.
It is also just a begining.
I would like to have a full graph descrition of the all chain from a script from data to clean spikes.
The multi segment is really a strong change in the object model but we need it.

I think we somthing like 6 month before having the full spikeinterface ported to this and maybe 6 other month to stabilize and debug. The 2 API will live togother for long time I think (unfortunatly).

I go back to NWB in a moment on how to port  from spikeextractor to this new API the old  code for writting into NWB.

PR #557 (2021-02-10): Fix klusta
- fix for Klusta when spikes are not identified in specific channel groups fix #554 
- a fix for blackrock that I forgot to commit before got bundled on it... since it's a small thing, I hope that's ok? fix #543
closed 2021-02-10 by alejoe91