SpikeInterface/spikeextractors #546: Writing SubRecordingExtractor to Nwb file using NwbRecordingExtractor.write_recording

khl02007, 2021-01-23:
@alejoe91 I noticed that in the absence of metadata input, using `NwbRecordingExtractor.write_recording` to write a SubRecordingExtractor to an Nwb file gives an error. This seems to be related to the channel group - if it inherits the channel group from the parent extractor, then the electrodes do not get added.

```
>>> import spikeextractors as se
>>> se.__version__
'0.9.3'
>>> R = se.NwbRecordingExtractor('/stelmo/nwb/raw/despereaux20191125_.nwb', electrical_series_name = 'e-series')
>>> sub_R = se.SubRecordingExtractor(R, channel_ids = [24, 25, 26, 27], start_frame = 54417159, end_frame = 54417159+10000)
>>> sub_R.get_channel_groups()
array([26, 26, 26, 26])
>>> se.NwbRecordingExtractor.write_recording(recording = sub_R, save_path = 'test.nwb', overwrite=True, use_timestamps = True)
/home/kyu/miniconda3/envs/nwb_datajoint/lib/python3.8/site-packages/pynwb/file.py:753: UserWarning: Date is missing timezone information. Updating to local timezone.
  warn("Date is missing timezone information. Updating to local timezone.")
/home/kyu/repos/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py:580: UserWarning: No metadata was passed specifying the electrode group for electrode 24, and the internal recording channel group was assigned a value (26) outside the indices of the electrode groups in the nwbfile! Electrode will not be added.
  warnings.warn("No metadata was passed specifying the electrode group for "
/home/kyu/repos/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py:580: UserWarning: No metadata was passed specifying the electrode group for electrode 25, and the internal recording channel group was assigned a value (26) outside the indices of the electrode groups in the nwbfile! Electrode will not be added.
  warnings.warn("No metadata was passed specifying the electrode group for "
/home/kyu/repos/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py:580: UserWarning: No metadata was passed specifying the electrode group for electrode 26, and the internal recording channel group was assigned a value (26) outside the indices of the electrode groups in the nwbfile! Electrode will not be added.
  warnings.warn("No metadata was passed specifying the electrode group for "
/home/kyu/repos/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py:580: UserWarning: No metadata was passed specifying the electrode group for electrode 27, and the internal recording channel group was assigned a value (26) outside the indices of the electrode groups in the nwbfile! Electrode will not be added.
  warnings.warn("No metadata was passed specifying the electrode group for "
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-4-8d9d8290787c> in <module>
      1 # sub_R.set_channel_groups([0]*4, channel_ids = [24,25,26,27])
----> 2 se.NwbRecordingExtractor.write_recording(recording = sub_R,
      3                                          save_path = 'test.nwb', overwrite=True, use_timestamps = True)

~/repos/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py in write_recording(recording, save_path, overwrite, nwbfile, use_timestamps, metadata, write_as_lfp)
    933                     nwbfile = NWBFile(**nwbfile_kwargs)
    934 
--> 935                 se.NwbRecordingExtractor.add_all_to_nwbfile(
    936                     recording=recording,
    937                     nwbfile=nwbfile,

~/repos/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py in add_all_to_nwbfile(recording, nwbfile, use_timestamps, metadata, write_as_lfp)
    833 
    834         # Add electrodes
--> 835         se.NwbRecordingExtractor.add_electrodes(
    836             recording=recording,
    837             nwbfile=nwbfile,

~/repos/spikeextractors/spikeextractors/extractors/nwbextractors/nwbextractors.py in add_electrodes(recording, nwbfile, metadata)
    588         # property 'gain' should not be in the NWB electrodes_table
    589         # property 'brain_area' of RX channels corresponds to 'location' of NWB electrodes
--> 590         channel_prop_names = set(recording.get_shared_channel_property_names()) - set(nwbfile.electrodes.colnames) \
    591             - set(['gain', 'location'])
    592         for channel_prop_name in channel_prop_names:

AttributeError: 'NoneType' object has no attribute 'colnames'
```

If I do `sub_R.set_channel_groups([0]*4, channel_ids = [24,25,26,27])` before calling `se.NwbRecordingExtractor.write_recording` there is no error. Seems to me like `write_recording` should preserve the channel group in the SubRE.

alejoe91, 2021-01-23:
Hi @khl02007 

Thanks for the report. Yeah looks like a bug. @CodyCBakerPhD  can you look into this one when you get a chance?

CodyCBakerPhD, 2021-01-23:
@alejoe91 Sure thing, on it!

CodyCBakerPhD, 2021-01-23:
@khl02007 I've replicated and explored this issue - it is, regrettably, an intended outcome of the use case for the electrode groups, though it does stem from a mapping conflict we've had for some time within spikeextractors.

The error occurs because of the above warnings resulting in no actual electrodes being written to the NWBFile, as they could not reconcile the electrode group membership automatically. The quick fix is to pass a metadata dictionary of the form

```
metadata = dict(
    Ecephys=dict(
        Device=dict(),
        ElectrodeGroup=dict(),
        Electrodes=[dict(
            name='group_name',
            description="My description.",
            data=["26"]*4
        )],
        ElectricalSeries=dict()
    )
)
```
into the final call to the `NwbRecordingExtractor.write_recording(..., metadata=metadata)`, and that should resolve the problem for this case.

To explain why this issue occurs: spikeextractors only allows integer-valued channel groups, whereas Nwb allows them to be string-valued. We thought we [did our best to try to reconcile the mapping](https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/nwbextractors/nwbextractors.py#L569-L578) for cases like this, which occurs because you have (I assume) 26 valid channel groups in the first RecordingExtractor, but the SubRecording would only  have one (albeit with an group 'index' of 26). The original code would therefore expect that single electrode group to either be re-indexed to zero, or otherwise map to the 26th index of the electrode group table in the NWBFile (which, of course, only has one index here with name value of "26"). The metadata fix above essentially tells the NWB writer that the channel group (26) corresponds to electrode group ("26") which had been written correctly.

@alejoe91 This type of use case does imply that metadata is not entirely optional as indicated in the docstrings, since the code cannot run to completion without it. We can either (a) replace or enhance the warning at [these lines](https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/nwbextractors/nwbextractors.py#L579-L584) with a more informative assertion on how to correct the use (since not writing ANY electrodes at all is certainly a problem), (b) attempt something fancy with trying to match string names to integers in the case they only contain digits, or (c) bigger refactor extending spikeextractors channel groups to string values. Any preference between these?

CodyCBakerPhD, 2021-01-26:
@khl02007 After some internal discussion we are going to extend the way electrodes form their group mapping to try to resolve this very issue in a way that doesn't require the metadata to be passed manually. Hope to get that fixed for you in the next couple of days!

khl02007, 2021-01-26:
@CodyCBakerPhD  Great, thank you!

khl02007, 2021-02-16:
This was fixed by https://github.com/SpikeInterface/spikeextractors/pull/552. Thanks all.

closed 2021-02-16 by khl02007