SpikeInterface/spikeextractors #550: Tridesclous extractor gets spikes only from first segment

remi-pr, 2021-01-28:
Hi everyone,

Great work on the spikeinterface toolbox, this is extremely helpful, and clearly working toward better neuroscience.

While using the spikeextractor for tridesclous I stumbled upon this [line](https://github.com/SpikeInterface/spikeextractors/blob/3ee107b7381e38b43c72f6a5dfc651ad3f1fb44d/spikeextractors/extractors/tridescloussortingextractor/tridescloussortingextractor.py#L40), which is making the extractor return only the spikes from the first segment.

I guess one could do something like this (careful, this is ugly):
`        self._all_spikes = np.hstack([dataio.get_spikes(seg_num=seg_ix, chan_grp=self.chan_grp,
                                                        i_start=None, i_stop=None).copy()
                                      for seg_ix in range(dataio.nb_segment)])`

to make sure to get everything. I am not suggesting this as a solution, just to highlight the issue. 
Let me know if I can be of any help.
Cheers.

alejoe91, 2021-01-28:
Thanks @remi-pr 

I'll let @samuelgarcia handle this :)

samuelgarcia, 2021-01-28:
The entire spikeinterface familly in mono segment.
This is why tridesclous extractor is supposed to extract only one segment.
Otherwise it means that the tridesclous folder have been generated ousite spikeinterface tools.

In spikeinterface the multisegment is handled whith MultiRecordingTimeExtractor which is totally horrible because  before any sorter a concatenate of all trace file is dumped to a big binary file.

I am in process of refactoring spikeinterface to handle this multi segment stuff.
It is a big channlenge and now I quite alone for that.
When done in several month I think then spikeinterface will be able to load a multisegment tridesclous folder.