SpikeInterface/spikeextractors #553: shared_channel_property_names issue: CacheRecordingExtractor of a MultiRecordingChannelExtractor

ferchaure, 2021-02-03:
Hi Guys, I'm testing the effect of whitening in subgroups of channels following this workflow:

1.  I load the recording and add multiple properties to each channel like channelD, probe_number, etc
2. I apply st.preprocessing.whiten to groups of channels using get_sub_extractors_by_property  (for example by probe number) and then combine the sub_extractors in one recording using MultiRecordingChannelExtractor.
3. Finally, I apply mountainsort with the argument grouping_property equals to 'channelD',  'probe_number' or any other property. 

That works fine but then I tried to use a CacheRecordingExtractor between point 2 and 3. I found that the CacheRecordingExtractor loose all the shared_channel_property_names and only keeps ['group', 'location']. Is it a bug or am I just overusing the spikeinterface classes with the intricate workflow?

colehurwitz, 2021-02-03:
Thanks for letting us know about this! I would say that it this is unintended behaviour and most likely a bug. I will look into it!

colehurwitz, 2021-02-04:
Could you actually post a code snippet for me to look into? @ferchaure

ferchaure, 2021-02-05:
Hi Cole,  Below a toy example with the issue 

```
import spiketoolkit as st
import spikeextractors as se
import numpy as np

num_channels = 2
sampling_frequency = 30000  # in Hz
duration = 1
timeseries = np.random.normal(0, 1, (num_channels, int(sampling_frequency * duration)))
geom = np.zeros((num_channels, 2))
recording = se.NumpyRecordingExtractor(timeseries=timeseries, geom=geom, sampling_frequency=sampling_frequency)

for ch in range(num_channels):
    recording.set_channel_property(property_name='probe_number', value=2, channel_id=ch)

rlist = [st.preprocessing.whiten(r) for r in recording.get_sub_extractors_by_property('probe_number')]
new_recording = se.MultiRecordingChannelExtractor(rlist)
new_recording_c = se.CacheRecordingExtractor(new_recording)

new_recording.get_shared_channel_property_names()
#['group', 'location', 'probe_number']

new_recording_c.get_shared_channel_property_names()
# ['group', 'location']
```

colehurwitz, 2021-02-05:
Thanks! Looking into this now

colehurwitz, 2021-02-05:
Ok, so it looks like a nasty bug where the `MultiChannelRecordingExtractor` has not yet been updated to for the new system of 'key_properties' and 'properties' being handled separately internally.

I will need a little bit of time to figure out the best way to update `MultiChannelRecordingExtractor`.

colehurwitz, 2021-02-05:
Ok, I have a solution in mind. Will push a change in the near future.

PR #555 (2021-02-10): property fixes multichannel
Fixes #553 

Quick explanation: `MultiRecordingChannelExtractor` used to fetch properties from each internal extractor, as a result, the `._properties` dictionary was empty. As `copy_channel_properties` was sometimes deepcopying the `._properties` dictionary, this would cause properties to not be inherited by the `CacheRecordingExtractor` when wrapping a `MultiRecordingChannelExtractor`.

Fix: `MultiRecordingChannelExtractor` now fills the `._properties` dictionary with all of the internal extractor properties and then does not reference the internal extractor prope [... cut]
closed 2021-02-10 by colehurwitz