SpikeInterface/spikeextractors #554: Error on run_klusta

luiztauffer, 2021-02-05:
I'm getting this error when running klusta sorting. It only happens when I run it with all channels (a "full" recording). If i run it with a subrecording, using just a small subset of channels, the sorting runs well.

```
sorting = ss.run_klusta(
    recording=rec_to_sort, 
    grouping_property='group',
    output_folder=dir_spikeinterface / "si_output",
)
```

```
WARNING! No channel location given. Add dummy location.
WARNING! No channel location given. Add dummy location.
...
WARNING! No channel location given. Add dummy location.
WARNING! No channel location given. Add dummy location.
RUNNING SHELL SCRIPT: /media/luiz/storage/taufferconsulting/client_ben/project_movshon/data/spikeinterface/si_output/0/run_klusta.sh
RUNNING SHELL SCRIPT: /media/luiz/storage/taufferconsulting/client_ben/project_movshon/data/spikeinterface/si_output/1/run_klusta.sh
...
RUNNING SHELL SCRIPT: /media/luiz/storage/taufferconsulting/client_ben/project_movshon/data/spikeinterface/si_output/190/run_klusta.sh
RUNNING SHELL SCRIPT: /media/luiz/storage/taufferconsulting/client_ben/project_movshon/data/spikeinterface/si_output/191/run_klusta.sh
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-17-2ac241f87e3b> in <module>
      7     recording=rec_to_sort,
      8     grouping_property='group',
----> 9     output_folder=dir_spikeinterface / "si_output",
     10 )

~/anaconda3/envs/env_klusta/lib/python3.6/site-packages/spikesorters/sorterlist.py in run_klusta(*args, **kwargs)
    263         The spike sorted data
    264     """
--> 265     return run_sorter('klusta', *args, **kwargs)
    266 
    267 

~/anaconda3/envs/env_klusta/lib/python3.6/site-packages/spikesorters/sorterlist.py in run_sorter(sorter_name_or_class, recording, output_folder, delete_output_folder, grouping_property, parallel, verbose, raise_error, n_jobs, joblib_backend, **params)
     89     sorter.set_params(**params)
     90     sorter.run(raise_error=raise_error, parallel=parallel, n_jobs=n_jobs, joblib_backend=joblib_backend)
---> 91     sortingextractor = sorter.get_result()
     92 
     93     return sortingextractor

~/anaconda3/envs/env_klusta/lib/python3.6/site-packages/spikesorters/basesorter.py in get_result(self)
    234 
    235     def get_result(self):
--> 236         sorting_list = self.get_result_list()
    237         if len(sorting_list) == 1:
    238             sorting = sorting_list[0]

~/anaconda3/envs/env_klusta/lib/python3.6/site-packages/spikesorters/basesorter.py in get_result_list(self)
    229         sorting_list = []
    230         for i, _ in enumerate(self.recording_list):
--> 231             sorting = self.get_result_from_folder(self.output_folders[i])
    232             sorting_list.append(sorting)
    233         return sorting_list

~/anaconda3/envs/env_klusta/lib/python3.6/site-packages/spikesorters/klusta/klusta.py in get_result_from_folder(output_folder)
    157     @staticmethod
    158     def get_result_from_folder(output_folder):
--> 159         sorting = se.KlustaSortingExtractor(file_or_folder_path=Path(output_folder) / 'recording.kwik')
    160         return sorting

/media/luiz/storage/Github/spikeextractors/spikeextractors/extractors/klustaextractors/klustaextractors.py in __init__(self, file_or_folder_path, exclude_cluster_groups)
    103 
    104         for channel_group in kf_reader.get('/channel_groups'):
--> 105             chan_cluster_id_arr = kf_reader.get(f'/channel_groups/{channel_group}/spikes/clusters/main')[()]
    106             chan_cluster_times_arr = kf_reader.get(f'/channel_groups/{channel_group}/spikes/time_samples')[()]
    107             chan_cluster_ids = np.unique(chan_cluster_id_arr)  # if clusters were merged in gui,

TypeError: 'NoneType' object is not subscriptable
```

all folders and files under `si_output` seem to be created correctly...

luiztauffer, 2021-02-05:
my bad... just realized there's no groups in the specific recording I was testing with in this case

luiztauffer, 2021-02-05:
my bad v2: there ARE groups, they're individual groups so the sorter runs independently... the issue is alive

alejoe91, 2021-02-05:
@luiztauffer the sorting seems to have run fine. Can you look into the `KlustaSortingExtractor` to see if you find a quick fix for this?

luiztauffer, 2021-02-05:
seems to be a problem with the `recording.kwik` file for one specific channel (possibly more)

The KlustaSortingExtractor has this code:
https://github.com/SpikeInterface/spikeextractors/blob/01564740d4b476ec2a16bc2b25119a809897860b/spikeextractors/extractors/klustaextractors/klustaextractors.py#L105

But looking at the file where the problem first happens, we see that the h5 path looks different:
```
kf_reader = h5py.File('/path_to/si_output/17/recording.kwik', 'r')
a = kf_reader.get('/channel_groups/0/')
a.keys()
```
```
<KeysViewHDF5 ['channels']>
```
it doesn't have the key `spikes`

luiztauffer, 2021-02-05:
I'm not sure how to proceed from there:
- is it a fault of the sorter? Every `recording.kwik` file should have that `spikes` key over there? Maybe no spikes were found on that channel?
- It should be expected that sometimes these keys are not present? In what cases?

alejoe91, 2021-02-05:
It might be that no spikes were found on that channel. Do you see the `spikes` key on other channels?

luiztauffer, 2021-02-08:
> It might be that no spikes were found on that channel. Do you see the spikes key on other channels?

I think it is the case indeed. Running it for longer pieces of data is does find spikes for that channel, and in general the h5 file returns more keys to that path:
```
<KeysViewHDF5 ['channels', 'cluster_groups', 'clusters', 'spikes']>
```

But if no spikes are found for a given channel, this should be handled better than throwing an error, right?

alejoe91, 2021-02-08:
> > It might be that no spikes were found on that channel. Do you see the spikes key on other channels?
> 
> I think it is the case indeed. Running it for longer pieces of data is does find spikes for that channel, and in general the h5 file returns more keys to that path:
> 
> ```
> <KeysViewHDF5 ['channels', 'cluster_groups', 'clusters', 'spikes']>
> ```
> 
> The error however will show up eventually in other channels, whenever it doesn't contain detected spikes.This should be handled better than throwing an error, right?

Yes it should be handled! I guess I never encountered it before.. can you add a check there?

PR #557 (2021-02-10): Fix klusta
- fix for Klusta when spikes are not identified in specific channel groups fix #554 
- a fix for blackrock that I forgot to commit before got bundled on it... since it's a small thing, I hope that's ok? fix #543
closed 2021-02-10 by alejoe91