SpikeInterface/spikeextractors #564: ValueError: channel_ids and locations must have same length

gmaggi, 2021-02-10:
Hello,

I got two issues with the master extractor:

1. spikeextractors/recordingextractor.py", line 237, in set_channel_locations
    raise ValueError("channel_ids and locations must have same length")
ValueError: channel_ids and locations must have same length

is this because of the inputs  that I am passing? In a previous version of spikeextractor (0.9.0), this error did not pop up

2.- there is a type here: https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/spikeglxrecordingextractor/spikeglxrecordingextractor.py#L25 (nameor)

Thanks!

alejoe91, 2021-02-10:
Hi Giuliano,

How do you get the error? Can you paste your code?

Thanks for finding the typo! Fixing now :) note that we are udating our tests to use example files so we expect that these kind of bugs will be easier to spot in the future.

Cheers
Alessio

gmaggi, 2021-02-10:
Thanks for the prompt response!!
the code is:

```
recordings=[]
for inputf in inputFiles:
     recording = se.SpikeGLXRecordingExtractor(inputf)
     recordings.append(recording)

subrecording = se.MultiRecordingTimeExtractor(recordings)
```

Below the full error:

`Traceback (most recent call last):
  File "/export/opt/computation_packages/bin/spike_interface_pipeline.py", line 305, in <module>
    recording = run_extractor()
  File "/export/opt/computation_packages/bin/spike_interface_pipeline.py", line 183, in run_extractor
    recording = se.SpikeGLXRecordingExtractor(inputf)
  File "/export/opt/anaconda2/envs/spikeinterface0.11.0_10.02.2021/lib/python3.7/site-packages/spikeextractors/extractors/spikeglxrecordingextractor/spikeglxrecordingextractor.py", line 85, in __init__
    self.set_channel_locations(locations)
  File "/export/opt/anaconda2/envs/spikeinterface0.11.0_10.02.2021/lib/python3.7/site-packages/spikeextractors/recordingextractor.py", line 237, in set_channel_locations
    raise ValueError("channel_ids and locations must have same length")
ValueError: channel_ids and locations must have same length`

alejoe91, 2021-02-10:
@gmaggi apparently it found a different number of locations than number of channels, and that triggers (correctly) the Error. Does it happen for all recordings? 

Do you mind sharing the recording that raises the error with me? You can send me an email and we can arrange it

gmaggi, 2021-02-10:
I've tested it with 4 files, and the error is the same.
I just sent you an email.
Thanks,
Giuliano.

alejoe91, 2021-02-11:
Hi Giuliano,

Turns out it was a quick fix: https://github.com/SpikeInterface/spikeextractors/compare/spikeglx_channels_fix?expand=1#

The error was due to a mismatch in the naming when the file names have `imec` instead of `imec0`.

Let me know if it fixes the issue for you.

Cheers
Alessio

closed 2021-02-11 by alejoe91