SpikeInterface/spikeextractors #574: trouble uploading open ephys recording

shirquinn, 2021-02-17:
Hi,
Iv'e recorded data from 32ch probe using the open ephys v0.5.3.1, raw binary format, with ADC channels (1 ADC used for photodiode)
when uploading the data i get an error:
>>> recording.get_channel_ids()
.dat:  continuous.dat
Estimated samples:  44648947  Numchan:  40
Warning: timestamps and nsamples are different!

Have anyone else encountered this problem and can assist?

alejoe91, 2021-02-17:
Hi, it's not an error but a warning. It means that the timestamps saved by the openephys are not the same as the estimates samples. Spikeextractors doesn't use the native timestamps so no worries, it won't affect the processing

shirquinn, 2021-02-17:
If i need to extract epochs according to TTL events, it won't be a problem?

alejoe91, 2021-02-17:
Usually the mismatch can be one or two frames, so I don't think it's a problem ;)

shirquinn, 2021-02-17:
This didn't happen when recording with an older version of open ephys (one without a recording node)
does it have to do with the way the recording node saves the information? 
"Currently, the synchronizer will only generate outputs when using the Binary data format for recording. The result is a synchronized_timestamps.npy file that contains a double (float64) timestamp for each int64 timestamp found in the original, unsynchronized timestamps.npy file for each data source."
I'm sorry for the stupid question- how can I access the estimates samples to compare?

alejoe91, 2021-02-17:
HI we can ask @jsiegle to elucidate us :)

You can check the number of timestamps by loading the timestamps.npy file in the continuous folder of your open ephys folder. The number of estimated samples are printed (44648947 in your case), and they are estimated from the number of channels and the size of the binary file.

shirquinn, 2021-02-17:
I've realized that if i de-select channles from the recording node it produces this warning( i had 333 ms extra in timestamps.npy than the estimated samples), i have no idea why this happens, but i tried to record without messing with the recording node and it doesn't produce this error

closed 2021-02-17 by shirquinn

jsiegle, 2021-02-18:
Hi @shirquinn, what version of the GUI were you using when this happened? If you can replicate the bug, please post an issue [here](https://github.com/open-ephys/plugin-GUI/issues) and we'll try to get it fixed.