SpikeInterface/spikeextractors #577: pynwb unit test fails on hdmf

AndyMender, 2021-02-21:
I'm trying to build a `python-spikeextractors` package for Fedora and noticed that one of the unit tests is failing:
```python
+ /usr/bin/python3 -m unittest discover
E
======================================================================
ERROR: spikeextractors (unittest.loader._FailedTest)
----------------------------------------------------------------------
ImportError: Failed to import test module: spikeextractors
Traceback (most recent call last):
  File "/usr/lib64/python3.9/unittest/loader.py", line 470, in _find_test_path
    package = self._get_module_from_name(name)
  File "/usr/lib64/python3.9/unittest/loader.py", line 377, in _get_module_from_name
    __import__(name)
  File "/builddir/build/BUILD/spikeextractors-0.9.0/spikeextractors/__init__.py", line 10, in <module>
    from .extractorlist import *
  File "/builddir/build/BUILD/spikeextractors-0.9.0/spikeextractors/extractorlist.py", line 11, in <module>
    from .extractors.nwbextractors.nwbextractors import NwbRecordingExtractor, NwbSortingExtractor
  File "/builddir/build/BUILD/spikeextractors-0.9.0/spikeextractors/extractors/nwbextractors/__init__.py", line 1, in <module>
    from .nwbextractors import NwbRecordingExtractor, NwbSortingExtractor
  File "/builddir/build/BUILD/spikeextractors-0.9.0/spikeextractors/extractors/nwbextractors/nwbextractors.py", line 14, in <module>
    import pynwb
  File "/usr/lib/python3.9/site-packages/pynwb/__init__.py", line 260, in <module>
    from . import io as __io  # noqa: F401,E402
  File "/usr/lib/python3.9/site-packages/pynwb/io/__init__.py", line 1, in <module>
    from . import base as __base
  File "/usr/lib/python3.9/site-packages/pynwb/io/base.py", line 1, in <module>
    from .core import NWBContainerMapper
  File "/usr/lib/python3.9/site-packages/pynwb/io/core.py", line 9, in <module>
    from pynwb.file import NWBFile
  File "/usr/lib/python3.9/site-packages/pynwb/file.py", line 13, in <module>
    from .base import TimeSeries, ProcessingModule
  File "/usr/lib/python3.9/site-packages/pynwb/base.py", line 9, in <module>
    from .core import NWBDataInterface, MultiContainerInterface, NWBData
  File "/usr/lib/python3.9/site-packages/pynwb/core.py", line 37, in <module>
    class NWBContainer(NWBMixin, Container):
  File "/usr/lib/python3.9/site-packages/pynwb/__init__.py", line 131, in _dec
    __TYPE_MAP.register_container_type(namespace, neurodata_type, cls)
  File "/usr/lib/python3.9/site-packages/hdmf/utils.py", line 561, in func_call
    return func(args[0], **pargs)
  File "/usr/lib/python3.9/site-packages/hdmf/build/manager.py", line 793, in register_container_type
    spec = self.__ns_catalog.get_spec(namespace, data_type)    # make sure the spec exists
  File "/usr/lib/python3.9/site-packages/hdmf/utils.py", line 561, in func_call
    return func(args[0], **pargs)
  File "/usr/lib/python3.9/site-packages/hdmf/spec/namespace.py", line 314, in get_spec
    raise KeyError("'%s' not a namespace" % namespace)
KeyError: "'core' not a namespace"
```
It might as well be an issue with `h5py`, rather than `spikeextractors`, though.

alejoe91, 2021-02-22:
Hi @AndyMender 

Thanks for building a Fedora package for spikeextractors!

Can you give it a try with `h5py==2.10`?

@CodyCBakerPhD @bendichter have you seen this before?

bendichter, 2021-02-22:
I don't think python 3.9 is supported yet

bendichter, 2021-02-22:
@AndyMender did you `git clone` hdmf? If so, that may be the issue. hdmf contains a submodule that needs to also be checked out. To do properly install from git, you need to run:

```bash
$ git clone --recurse-submodules git@github.com:hdmf-dev/hdmf.git
$ cd hdmf
$ pip install -r requirements.txt
$ pip install -e .
```

sanjayankur31, 2021-02-27:
Hello,

In Fedora, hdmf is built from the GitHub release tar, which seems to include the hdmf-common-schma files too.

https://src.fedoraproject.org/rpms/python-hdmf/blob/rawhide/f/python-hdmf.spec#_19

The build seems to be fine too. Here's the log for the latest one:
https://kojipkgs.fedoraproject.org//packages/python-hdmf/2.1.0/2.fc35/data/logs/noarch/build.log

We're at version 2.1.0 in Fedora, since pynwb 1.4.0 requires it.