SpikeInterface/spikeextractors #583: handle imec1.ap file

gmaggi, 2021-02-25:
Hello,

One of the spikeinterface-users in my institution is using two probes, every dataset contains a imec0.ap.bin and imec1.ap.bin.

Would it be possible to handle imec1 too?. She currently gets an assert-error from : 

https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/spikeglxrecordingextractor/spikeglxrecordingextractor.py#L38

Thanks,
Giuliano.

alejoe91, 2021-02-25:
Hi Giuliano,

We would love to improve our support on the Neuropixels side. Unfortunately, we don't know all the available versions/configurations and we are trying to figure it out. Since you guys are experts on this, would you like to help us out to cover all Neuropixels possibilities?

Let us know!

MarineChap, 2021-02-25:
Hello Alejo, 

I am not sure to know every versions/configurations either. @gmaggi  Maybe you could ask to Cagatay ?  

But for now, you could just replace your `imec0.ap` assert by a regex check looking like this : ` imec[0-9]?.ap`  and that would already cover multi-probe configurations. 
Also, it seems that it was working before. What happened that you decide to enforce a strict path style like this ? Maybe a little more flexibility would give you less maintainability problems. 

We have a lab meeting this morning. I can check for other config existing. 

(ps: Just in case, my intervention here was not clear Giuliano and I are both part of Nerf.)

Have a good day.

alejoe91, 2021-02-25:
Thanks @MarineChap 

Could you make a PR that encompasses all your neuropixels configurations/use cases? :)

samuelgarcia, 2021-02-25:
neo now support spikeglx.
the neo implementation support nativelly the multi stream stuuf  (imecX and nidacq)
Maybe we could write a wrapper on top of it rather than patching the actual implementation.

MarineChap, 2021-02-25:
I never used neo myself.  
What is the advantage to use it instead of directly spikeglxextractor ?  
Are you advised us to change in the long term for the neo extractor ? 

The patching is very small for the actual spikeglx implementation. And as all path seems to be infer from the input path, nothing else should be change right ?

samuelgarcia, 2021-02-25:
On the long term : yes my plan is to move all extractor on top of neo.
On the short term : If your patch is simple just do it.

You should have a look to neo and related tools (elephant, ephyviewer, ...). (no personal promotion here of course)

For spike sorting pipenline, spikeextractor is your unique choice now. Some neo.io are wrapper into spikeextractors.
Many extractor are redundant to neo implementation.

MarineChap, 2021-02-25:
Interesting. I will take a look once I have a little more time. Thanks for the update. 
What is long term ? few months or few years ? 

I will open a PR but I don't have the resource right now to test a full spike sorting pipeline to be sure it is not breaking elsewhere.

samuelgarcia, 2021-02-25:
I have started this : https://github.com/SpikeInterface/spikeinterface/tree/big_refactoring

I my mind this is something like 6 month.
If i get some help....

PR #584 (2021-02-26): Modified assertion check to accept all imecX.lf or imecX.ap file as input
Accept  now
- imec.ap.bin
- imec.lf.bin
- imecX.lf.bin      with X the recording number
- imecX.ap.bin

close #583 

@gmaggi  Do you think you could checkout on this branch on the cluster please ? I want to verify that the pipeline is not breaking later with the imec1 file and also Hanna needs to send her jobs as soon as possible. 

@alejoe91 , @samuelgarcia    should i add some tests to keep track of the allowed filenames ?
closed 2021-02-26 by samuelgarcia