SpikeInterface/spikeextractors #587: 'No acquisitions found in the .nwb file.'

AlanoudAlsubhi, 2021-03-01:
I am trying to follow the tutorial you have online on SpikeInterface. I am trying to use “NwbRecordingExtractor” to load my raw data which is on nwb format. I am attaching the error message that I keep getting. I am also attaching a print of my nwb file. 

[image]
[image]

CodyCBakerPhD, 2021-03-01:
How strange, the reader is supposed to infer the name of the electrical series automatically if not set.

If you specify the argument `electrical_series_name='continuous_signals'` in your call to `se.NwbRecordingExtractor`, does this resolve the issue?

alejoe91, 2021-03-01:
Hi @AlanoudAlsubhi 

I'll forward this to @CodyCBakerPhD @luiztauffer and @bendichter

alejoe91, 2021-03-01:
@CodyCBakerPhD was faster than me :)

AlanoudAlsubhi, 2021-03-01:
Hi Thanks for your response. 
I am just getting started with all this, and I am trying to get familiar with it. 

Do you mean something like this? 
[image]

Because it still returns an error. 
"TypeError: Indexing elements must be in increasing order" 

Thanks

AlanoudAlsubhi, 2021-03-01:
It is not on DANDI yet, but I can share it somehow. Does it have to be DANDI? 

[image]

bendichter, 2021-03-01:
@CodyCBakerPhD It looks like the electrode ids are not in increasing order, and that is causing a problem because h5py requires that indexes specifying a subregion of a dataset must be in increasing order. This is an annoying limitation of h5py. You could get around this by sorting the indices, selecting the subregion, and then re-shuffling it back into its original order.

bendichter, 2021-03-01:
I can provide a google drive that can be used to share the data if that would make this easier.

CodyCBakerPhD, 2021-03-01:
@AlanoudAlsubhi It doesn't have to be on DANDI to share, I was just wondering if this was work in progress or an attempt to use a more established dataset.

It looks like we solved the first issue of the acquisition names (will still try to figure out why the auto-naming didn't work, though...), but now as @bendichter indicated there's an issue I haven't seen before on the channel orderings (which are usually added in consecutive order). We can fix that pretty easily on the spikeextractors side, but it would still help for holistic debugging to have access to the data if you're willing to share the file temporarily.

I'll open up a branch for the channel indexing and try to get this resolved by tomorrow!

AlanoudAlsubhi, 2021-03-01:
Yes, absolutely. I can share a sample file on the google drive @bendichter.

bendichter, 2021-03-01:
@AlanoudAlsubhi what email should I send the invite to?

AlanoudAlsubhi, 2021-03-01:
alanoud.alsubhi@gmail.com

AlanoudAlsubhi, 2021-03-01:
File is uploaded.

CodyCBakerPhD, 2021-03-02:
@AlanoudAlsubhi I'm making some progress on this due to the data share (thanks for that!), and it lead to the discovery of an [issue with pynwb](https://github.com/NeurodataWithoutBorders/pynwb/issues/1342) that I'm trying to fix now as well to prevent this from happening in the future.

So the problem may have two parts to it:

1) electrode ids in the `nwbfile.electrodes` table should be unique (but pynwb let's you add them non-uniquely, which is the aforementioned bug), whereas you have them set to what is essentially their ID within their given electrode group (which is usually information attached as an extra custom column, much like you have for the other fields like `probe_id`). If you reconvert the data with these unique IDs, I hope that should resolve the issue.

If it doesn't, then 2) I noticed that the h5py ordering issue is only triggered for the DynamicTableRegion you've setup in this file (which shows up as an oddly-sized array of integers), and I'm having trouble reproducing that through our usual methods of making that region. Can you post the code used to create the electrode table for this NWBFile, and the code for linking the electrode region for the ElectricalSeries in the `nwbfile.acquisition`?

AlanoudAlsubhi, 2021-03-02:
Hi @CodyCBakerPhD, Thank you so much for your efforts. I am reaching out to [NeuroNexus](https://neuronexus.com/) for their input regarding these issues because our nwb files were converted using their [Allego Software](https://neuronexus.com/allego-software-downloads-page/). 

Since I didn't convert the data manually, am I still able to fix the electrode ids on my file?

CodyCBakerPhD, 2021-03-11:
@AlanoudAlsubhi Sorry for the delay in further responses to this issue - there is now a [PR up on spike interface](https://github.com/SpikeInterface/spikeextractors/pull/600) which can fix both minor issues discovered here.

Unfortunately, the greater issue of non-unique global electrode ids is a schema violating issue that would be a little hacky to resolve just in spike interface for this case. I now have a solution available to the [source bug in PyNWB](https://github.com/NeurodataWithoutBorders/pynwb/pull/1344), and we have reached out the NeuroNexus team to try to resolve the issue there as well.

As for a quick/temporary solution: with the Allegro software, are you able to save the files in any other format? If you are able to, check which formats it allows saving to and compare to the other extractor types supported here in `spikeextractors`, and hopefully there is some intersection between the two which can enable you to use the rich toolkit of SpikeInterface. You can also then use the `NwbRecordingExtractor.write_recording()` function to write your data from that intermediate format to a valid NWBFile structure for easier sharing and publication!

PR #600 (2021-03-13): fix issue 587
fix #587 

Part (a) of that issue dealt with auto-population of the electrical series name, which is now properly fixed.
Part (b) is the odd error that occurs, for example, if the electrode ids in the nwbfile.electrodes table were not added in increasing order - the original direct indexing of the electrical series region (which are indexes of ids in the electrodes table) through the table caused an error through h5py and is now fixed by list compression passing through a single index at a time.
Part (c) is unfortunately the fact that the electrode ids weren't unique to begin wit [... cut]
closed 2021-03-13 by alejoe91