SpikeInterface/spikeextractors #588: properties like mda_max_channel on multisortingextractor

ferchaure, 2021-03-02:
Hi, Guys

I have been sorting using grouping_property to sort a recording grouped by channel, probe, etc. I tried to save the result on a firings.mda file but the first column doesn't make sense, that is because the property mda_max_channel is related to the SubExtractorsRecording of each group. For example, if I run msort to each channel independly then all the mda_max_channel are 0 (only one channel in a group); however, the values should be the ch_id related to each unit. 

Maybe creating a special case for the method  get_unit_property on MultiSortingExtractor to correct the values for mda_max_channel  (and probably others similar properties)?  And adding a flag to the properties with values related to channel ids that could be changed after combining subextractors.

alejoe91, 2021-03-03:
Hi @ferchaure 

If you spike sort by group, in my opinion it makes sense that the `mda_max_channel` don't correspond, as other channels are not seen by each run of the sorters. What I recommend is running the `st.postptocessing.get_unit_max_channels(recoridng, multisorting)` to get the correct max channels values. This problem seems to be related specificallyt to the `mda_max_channel`, which is only used by mda-base sorters. In my opinion it's not general enough to correct for it and it would also be a bit convoluted to correct for. Again, I think that recomputing the max channels with SI would be a better approach

ferchaure, 2021-03-03:
ok. However st.postptocessing.get_unit_max_channels find the peak on all the channels not only on each subrecording... and that kind of breaks the meaning of saving sorting results on a  .mda file. Anyway, I can make my own mda writer that make that adjustments.

closed 2021-03-03 by ferchaure