SpikeInterface/spikeextractors #589: Memory (mapping) problems with the OpenEphysExtractor

eejd, 2021-03-02:
I have been trying to work with SpikeInterface to setup a processing pipeline for tetrode data. The data sets are 16 tetrodes (64 channels) recoded using an Open Ephys system and saved in the Open Ephys ('continuous') data format. Despite trying a number of variations based on bits of the tutorials, I have struggled to get data sets loaded and passed to a sorter. Currently, shorter sessions (e.g. 7.5GB) can barely make it through an extractor -> filter -> sorter script to KiloSort. During the loading of the session, I hit 70GB of virtual memory used(!?). a 15GB session can't even load half the data before hitting allocation errors:
`MemoryError: Unable to allocate 36.4 GiB for an array with shape (42, 116332544) and data type float64`
Interesting, digging into the extractor code, I noted that it's default behavior seemed to *not* be memory mapped:
`
def __init__(self, folder_path, experiment_id=0, recording_id=0, dtype='float'):
        assert HAVE_OE, self.installation_mesg
        assert dtype == 'int16' or 'float' in dtype, "'dtype' can be int16 (memory map) or 'float' (load into memory)"
`
However, the above memory allocation error was from the following call:
`
recording = se.OpenEphysRecordingExtractor(session_folder + session_open_ephys, dtype='int16')
recording_drive = recording.load_probe_file(session_folder + session_open_ephys + 'drive.prb')
`
I note that this error only occurred when a (delayed) load was caused by the loading of the probe configuration. This seems to imply that the probe loading converted the 'int16' to 'float64'?!

I have the following questions (whose answers I feel need to be placed prominently in the documentation):
1. How is the deferred loading managed by the extractors? (What functions cause data loads?)
2. How can you ensure that file loading is memory mapped?
3. How can I enforce a conversion to a file format that is better for memory mapping while retaining additional information? (i.e. the BinDatRecordingExtractor appears to not retain meta information from an extractor when you use `write_to_binary_dat_format`, but I don't see any information on how to save the data with metadata for sub recordings, conversion with metadata, etc.)
4. How can I load partial data if the original extractor has segmented files. (In this case, if I call SubRecordingExtractor on an OpenEphysExtractor with multiple 'continuous' files, it loads *all* the data rather than loading the requested channels.)
5. Am I just doing something blindly wrong?

I am happy to help sort these with some direction and make a pull request, or help test. This was run using the pip installs (versions below) except for the spikesorters, which is the latest GitHub pull to include KiloSort3.
`
Python 3.9.1

pyopenephys         1.0.7
spikecomparison     0.3.1
spikeextractors     0.9.3
spikefeatures       0.1.1
spikeinterface      0.11.0
spikemetrics        0.2.3
spiketoolkit        0.7.2
spikewidgets        0.5.1
`

alejoe91, 2021-03-03:
Hi @eejd 

If you are using the open ephys format (you save a bunch of continuous files per channel) unfortunately memory map is not supported, because the channels need to be concatenated together..if you are using the binary format, then this problem should not happen.

What you can do is to cache the initial extractor to a temporary binary file using:

`recording_cache = se.CacheRecordingExtractor(recording_drive, 'my-path.dat')`

This will create a binary version of the data that is memory mapped. You can then `del recording, recording_drive` to release memory and proceed with the pipeline.

samuelgarcia, 2021-03-03:
Maybe we could use a wrapping on top of neo implementation for this no ? neo.openEphysRawio (old format) is lazy.

eejd, 2021-03-03:
The problem I encountered seems to be that the dtype was pushed up from int16 (or float) to float64. This means that I can't even load the data to do the conversion. For example, if I use the CachRecordingExtractor, I had the same memory load and inability to save the temporary file. (I'd tried this from an example in another issue.) So I think there is some part of this process which is either not passing the self._dtype or for some other reasons up-converting the dtype. I have a lot of these recording sessions that I'll need to process, so I'd like to help debug this or help implement a solution. My goal would be to have a pipeline that is reasonably efficient, can recover if there is a problem (so I'd like to have the mmappable form saved along with whatever session metadata I'd need so I can re-load from there without re-processing), and follows the spikeinterface conventions. I think doing full reads of the continuous files from long sessions would be only possible for me even to convert currently if I can keep the data in int16 or do any conversions after getting the data written to a memmapped object. (I guess I need to confirm the internal dtype of the continuous file, but I understood it to be int16.) 

I see a few options I can purse:

1. Work out a way to load the data as int16, save to the CacheRecordingExtractor, but I'll need to somehow preserve gain, etc and be able to load the probe file after the data is cached to memmapped binary.
2. Work out a method that loads and concatenates the continuous files to disk directly (the is, reads the files and appends the binary data one file at a time to limit the maximum read, like chunk reading). (I guess this might be best discussed on the pyopenephys repo.)
3. Create a new RecordingExtractor type that is a numpy extractor that can preserve meta data from the OpenEphys session, and provide a mechanism to load safely in the open ephys extractor and save to this for future use.

I guess I'd like advice (and help debugging the dtype problem) as well as knowing if I should work on the classic or try to understand the neo extractors. I'm going to investigate the CacheRecodingExtractor to understand why it was giving me memory errors now.

eejd, 2021-03-03:
Update on CacheRecordingExtracator. I just re-ran with the above suggestions (and requesting dtype='int16':

```
recording_oe = se.OpenEphysRecordingExtractor(session_folder + session_open_ephys, dtype='int16')
recording = se.CacheRecordingExtractor(recording_oe, save_path=path.expanduser('local/'))
del recording_oe
```

The result was the same:

```
MemoryError: Unable to allocate 36.4 GiB for an array with shape (42, 116332544) and data type float64
```

And specifically, shows that it was trying to allocate in float64

alejoe91, 2021-03-03:
Hi @eejd 

With this [PR](https://github.com/SpikeInterface/spikeextractors/pulls) (that we'll merge in a few days), I just double checked that you can get the traces of the `openephys` format in int16. We added a `return_scaled` arg to the `get_traces`, which also gets propagated to the cache extractor:

`rec_cache = se.CacheRecordingExtractor(rec_drive, return_scaled=False)`

So if you install from trat branch (and upgrade `pyopenephys`), you should be able to load and cache `int16` values. Alternatively, as Sam suggested, we could use NEO for the old open ephys format.

eejd, 2021-03-03:
I think there is a deeper problem, at least in the pyopenephys 1.0.7 required for the current master branch. In [the loadContinuous](https://github.com/CINPLA/pyopenephys/blob/1.0.7/pyopenephys/openephys_tools.py) helper function there is an argument for the dtype (float vs int16).

`def loadContinuous(filepath, dtype=float):`

However in the [core module](https://github.com/CINPLA/pyopenephys/blob/1.0.7/pyopenephys/core.py) it's called without:

`sig = loadContinuous(fullpath)`

I have modified (forcibly, to check) the call to `loadContinuous(fullpath, dtype=np.int16)` and the data files load without problem and with expected memory use. What I don't know is why this call (in core.py:Recording:_read_analog_signals) doesn't have access to a requested dtype.

It seems this [PR](https://github.com/CINPLA/pyopenephys/pull/20) is also relevant. Interestingly, looking at the pyopenephys code in the master, it seems that despite the default now being int16, it still doesn't pass through the dtype when you load via the top level AnalogSignal object, so that if you do want to have a dtype passed correctly from the SikeInterface extractor, you will now get int16 always! And I don't see any saved dtype in core.py, suggesting that the pyopenephys objects don't support requesting/storing their type, which seems like something that should be improved for SpikeInterface compatibility. (And generally, to know what the underlying data type is.)

alejoe91, 2021-03-03:
Ah yes sorry that is fixed in `pyopenephys>1.1`. I didn't connect it to that issue immediately, thanks for bringing it up. Can you try with the `return_scaled` branch and pyopenephys master if the issue is fixed? I haven't used the OpenEphys format much and I'm sure there's room for improvement. 

Right now our plan is to make a new release with these scaling issues fixed, and later base the OpenEphys extractors on NEO.

eejd, 2021-03-03:
Happy to help! I'll run that test (as soon as my script finishes pushing through kilosort which it finally is running). The `return_scaled=False` preserves the int16 throughout? I was curious if there is a reason to convert to a float32 or int32 post ingestion for any of the preprocessing or sorting downstream.

I think based on my experience, if the next release is not going to move to the NEO extractors (assuming they also solve this problem), anyone who might want to process Open Ephys 'continuous' files would need this fixed to be able make any real use of SI.

alejoe91, 2021-03-03:
Yeah unfortunately the new release is not going to include the NEO extractor. 
However, you are very welcome to contribute an `OpenEphysLegacyRecordingExtractor` based on neo! it's quite simple and we can merge it to amster as soon as it's done, so at least the master branch is fixed :) Take a look at some of the neo-based extractors here: https://github.com/SpikeInterface/spikeextractors/tree/master/spikeextractors/extractors/neoextractors

Let me know if that could work! 
Cheers
Alessio

eejd, 2021-03-04:
@alejoe91 I pulled the `return_scaled` branch and updated pyopenephys to `master`. I just tried running as you'd suggested, but it immediately threw an error as `return_scaled` was an unexpected keyword. I looked at the PR and the code and I see the support in `get_traces` and related functions, but none of the extractors seem to have them as init args (or properties). I don't know if this means I pulled the wrong branch or if it was never implemented at that level. This would have to be added to all the Extractors? (Do you have a developer guide or readme?)

alejoe91, 2021-03-04:
Sorry, I still have to add the `return_scaled` option to the `CacheRecordingExtractor` (give me a couple of days). But at least you should be able to `get_traces` in `int16` from the `OpenEphysRecordingExtractor`

eejd, 2021-03-04:
I'll do a first pass at implementing and testing the extractor with the keyword in the OE extractor and Cache. Should it be added to the base extractor class? If I have some guidance on style or dependency I'll make a PR if it works.

alejoe91, 2021-03-04:
@eejd I'm almost done with it! just need to test it properly and push it :) Thanks for the availability!

eejd, 2021-03-05:
FYI, `AttributeError: 'BandpassFilterRecording' object has no attribute 'has_unscaled'`, which I found when hack testing. @alejoe91, let me know when you push for testing.

alejoe91, 2021-03-05:
@eejd since this is an API change, we have a PR ready for spiketoolkit too: https://github.com/SpikeInterface/spiketoolkit/pull/440

alejoe91, 2021-03-18:
@eejd can you try with the latest releases?